esp8266/Arduino #309: Uploading sketch to Arduino via ESP8266

RussMathis, 2015-05-22:
How do I create a programmer that could be selected from the programmers in the "Aduino ESP8266 IDE" that would upload via sketches to the Arduino.

It would need an IP setting for the programmer?
The ESP8266 code would have to be a utility/library so I could still have code doing other things besides just and uploading.

Nice if it worked with any controller connected to the ESP8265 :)

marvinroger, 2015-05-25:
You have to turn the ESP8266 as an AVR ISP. Fortunately, this is pretty easy to do with a "native" Arduino board, using the [ArduinoISP.ino sketch](https://github.com/esp8266/Arduino/blob/esp8266/build/shared/examples/ArduinoISP/ArduinoISP.ino). The fact is this code won't work as is on the ESP8266, as the low level SPI interface is not the same. If you feel you can port the code, my bet is you would have to compare the [AVR SPI](https://github.com/esp8266/Arduino/tree/esp8266/hardware/arduino/avr/libraries/SPI) and [ESP8266 SPI](https://github.com/esp8266/Arduino/tree/esp8266/hardware/esp8266com/esp8266/libraries/SPI) and make the required changes.

RussMathis, 2015-05-26:
Ok, I've noticed the AVRDude has a Port feature, maybe some how porting that feature into "esptool" project would work?

igrr, 2015-08-05:
ESP8266AVRISP library by @kzyapkov merged in 50a1fce.

closed 2015-08-05 by igrr