esp8266/Arduino #310: SPIFFS @ficeto @iggr Very bad project

tontito, 2015-05-23:
I haven't seen anything so ridiculous.

remove spiffs because @ficeto wanted them.

now I don't follow this ridiculous project.

@ficeto if you're quoting, you need to specify

ficeto, 2015-05-23:
it is specified, he is quoted, will not quote him on files that no longer contain any of his work "that file was once used to hold somebody's code" you were also from the barking dogs, and I put 90% of the time in spiffs, so yes it should go. Now be happy no more of his or mine work on spiffs is there.

ficeto, 2015-05-23:
actually his is still there... bummer

tontito, 2015-05-23:
Your code import repository and change copyright and relase 

Copyright (c) 2015 Tontito. All rights reserved.
This file is part of the esp8266 core for Arduino environment.
 
This library is free software; you can redistribute it and/or
modify it under the terms of the GNU Lesser General Public
License as published by the Free Software Foundation; either
version 2.1 of the License, or (at your option) any later version.

how nice is it?

That`s shame.

ficeto, 2015-05-23:
that import was oh so mistake, So content was gone, If I would have started on clean, it would have taken me 5 times less time. Spiffs works out of the box, how to write to the flash is also well documented. And that is what I did DELETED HIS CODE and started clean.
Just so we are clear? Did you see the spiffs_esp8266 files? You see my name in them anywhere? Who's name you see?

tontito, 2015-05-23:
original write @pellepl

anakod license file 

https://github.com/anakod/Sming/blob/master/Sming/Services/SpifFS/LICENSE

he don`t remove copyrights

The MIT License (MIT)

Copyright (c) 2013-2015 Peter Andersson (pelleplutt1976<at>gmail.com)

this reaction was wrong.
only write Thank you Anakod.
that was it.

now anyway. good luck. Remove all pull and relax.

igrr, 2015-05-23:
But the licence file for spiffs library is there — https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/cores/esp8266/spiffs/LICENSE, so what's the issue?

ficeto, 2015-05-23:
but all of those are there? I really do not understand what you are stating...
what is left of his code has his name, spiffs is straight from github so those are there too.
You keep coming with reactions and yet have gone to look at the files...

tontito, 2015-05-23:
in fact, there was nothing.
remove the codes that was pretty funny.

ficeto, 2015-05-23:
nothing? coming from a person that comments on a commit diff? can't go a commit back and look?

ficeto, 2015-05-23:
it was removed because of people like you! now every damn SPIFFS implementation that come up for the ESP has to go through the BIG BOSS anakod... no thanks
@pellepl can look at the code (not that he should spend time on us) and I strongly believe that someone that actually writes code will be on my side. I have all commits forked on my own git server, so proof is here to stay. @anakod should revise his code, and maybe even get slapped for doing such a poor job, then wanting to be the man in every such implementation. He does not have a right to SPIFFS or any of the methods it exports.

tontito, 2015-05-23:
why so reacted.
you were working very constructive.
how a discussion like this came in such a beautiful project

ficeto, 2015-05-23:
you can see his remarks and my answers in the commit history... i di not start it. he did, then he shit on the forum, never showed his face to me to talk straight. Send you to do the talking...

tontito, 2015-05-23:
I don't know @anakod and not using Sming.

have a nice day.

ficeto, 2015-05-23:
read your comments, then look at his repo, then read the following file (the file that cause all of this) and tell me what is his:
```cpp
/* 
  FileSystem.cpp - SPIFS implementation for esp8266

  Copyright (c) 2015 Hristo Gochkov. All rights reserved.
  This file is part of the esp8266 core for Arduino environment.
 
  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
*/
#include "FileSystem.h"
#include "Arduino.h"
#include "spiffs/spiffs_esp8266.h"

// These addresses and sizes are defined in the linker script.
// For each flash memory size there is a linker script variant
// which sets spiffs location and size.
extern "C" uint32_t _SPIFFS_start;
extern "C" uint32_t _SPIFFS_end;
extern "C" uint32_t _SPIFFS_page;
extern "C" uint32_t _SPIFFS_block;

static s32_t api_spiffs_read(u32_t addr, u32_t size, u8_t *dst);
static s32_t api_spiffs_write(u32_t addr, u32_t size, u8_t *src);
static s32_t api_spiffs_erase(u32_t addr, u32_t size);

FSClass FS((uint32_t) &_SPIFFS_start, (uint32_t) &_SPIFFS_end, (uint32_t) &_SPIFFS_page, (uint32_t) &_SPIFFS_block, 4);

FSClass::FSClass(uint32_t beginAddress, uint32_t endAddress, uint32_t pageSize, uint32_t blockSize, uint32_t maxOpenFiles)
: _beginAddress(beginAddress)
, _endAddress(endAddress)
, _pageSize(pageSize)
, _blockSize(blockSize)
, _maxOpenFiles(maxOpenFiles)
, _fs({0})
{
}

int FSClass::_mountInternal(){
  if (_beginAddress == 0 || _beginAddress >= _endAddress){
    SPIFFS_API_DBG_E("Can't start file system, wrong address\r\n");
    return SPIFFS_ERR_NOT_CONFIGURED;
  }
  if(_pageSize == 0) _pageSize = 256;
  if(_blockSize == 0) _blockSize = 4096;
  
  spiffs_config cfg = {0};
  cfg.phys_addr = _beginAddress;
  cfg.phys_size = _endAddress - _beginAddress;
  cfg.phys_erase_block = INTERNAL_FLASH_SECTOR_SIZE;
  cfg.log_block_size = _blockSize;
  cfg.log_page_size = _pageSize;
  cfg.hal_read_f = api_spiffs_read;
  cfg.hal_write_f = api_spiffs_write;
  cfg.hal_erase_f = api_spiffs_erase;
  
  SPIFFS_API_DBG_V("FSClass::_mountInternal: start:%x, size:%d Kb\n", cfg.phys_addr, cfg.phys_size / 1024);

[... 278 lines cut ...]
}

static s32_t api_spiffs_read(u32_t addr, u32_t size, u8_t *dst){
  SPIFFS_API_DBG_V("api_spiffs_read: 0x%08x len: %u\n", addr, size);
  flashmem_read(dst, addr, size);
  return SPIFFS_OK;
}

static s32_t api_spiffs_write(u32_t addr, u32_t size, u8_t *src){
  SPIFFS_API_DBG_V("api_spiffs_write: 0x%08x len: %u\n", addr, size);
  flashmem_write(src, addr, size);
  return SPIFFS_OK;
}

static s32_t api_spiffs_erase(u32_t addr, u32_t size){
  SPIFFS_API_DBG_V("api_spiffs_erase: 0x%08x len: %u\n", addr, size);
  u32_t sect_first = flashmem_get_sector_of_address(addr);
  u32_t sect_last = flashmem_get_sector_of_address(addr+size);
  while( sect_first <= sect_last )
    if( !flashmem_erase_sector( sect_first ++ ) )
      return SPIFFS_ERR_INTERNAL;
  return SPIFFS_OK;
} 

```

igrr, 2015-05-23:
@tontito since you have opened this issue, how do you think it can be resolved?

closed 2015-05-23 by tontito