esp8266/Arduino #313: Get module's MAC address?

stozk, 2015-05-23:
Hi,

with the nodeMCU firmware I can read the modules preset MAC address.
Is this also possible with the esp8266 arduino sdk FW?

There are functions like "ESP.getChipId()" which return the chip's ID, but WiFi.macAddress(mac) can only be used to set an hard coded address, but I just want to read the modules original MAC.

Is this possible?

BR
Daniel

BJvDL, 2015-05-23:
String clientMac = "";
unsigned char mac[6];
WiFi.macAddress(mac);
clientMac += macToStr(mac);

String macToStr(const uint8_t* mac)
{
  String result;
  for (int i = 0; i < 6; ++i) {
    result += String(mac[i], 16);
    if (i < 5)
      result += ':';
  }
  return result;
}

stozk, 2015-05-23:
Ah nice thanks, thank you very much.
 So if I pass an empty char array it will return the preset MAC?

closed 2015-05-23 by stozk

BJvDL, 2015-05-23:
Yes. I use it for the ClientID with the MQTT connection

stozk, 2015-05-23:
I have simplified it a bit (imo ;-))

String getMacAddress() {
  byte mac[6];                     
  WiFi.macAddress(mac);
  String cMac = "";
  for (int i = 0; i < 6; ++i) {
  cMac += String(mac[i],HEX);
  if(i<5)
  cMac += "-";  
  }  
  cMac.toUpperCase();
  return cMac;
}

lvalero, 2016-03-29:
Leadnig "0" are missing, fixed here, thank you : 

String macToStr(const uint8_t* mac)
{
  String result;
  String atom;
  for (int i = 0; i < 6; ++i) {
    atom = String(mac[i], 16);
    if ( atom.length() < 2)
    {
      result += String ("0") + atom;  
    }
    else
    {
      result += atom;
    }
    if (i < 5)
      result += ':';
  }
  return result;
}

Humancell, 2016-03-29:
A slight variation ... I found the same thing:

String getMacAddress() {
  byte mac[6];

  WiFi.macAddress(mac);
  String cMac = "";
  for (int i = 0; i < 6; ++i) {
    if (mac[i]<0x10) {cMac += "0";}
    cMac += String(mac[i],HEX);
    if(i<5)
    cMac += "";   // put : or - if you want byte delimiters
  }
  cMac.toUpperCase();
  return cMac;
}

mtnbrit, 2016-03-29:
or simply

```
    WiFi.macAddress(mac);
    sprintf(mac_addr, "%02x:%02x:%02x:%02x:%02x:%02x", mac[0],mac[1],mac[2],mac[3],mac[4],mac[5]);
```

igrr, 2016-04-01:
Or simply `String s = WiFi.macAddress();`

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h#L60

lvalero, 2016-04-02:
i assume the function macToStr () is still needed when you scan the network WiFi.scanNetworks(); in order to retrieve BSSIDs

igrr, 2016-04-02:
Or you could just use WiFi::BSSIDstr(i)
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiScan.h#L49

lvalero, 2016-04-02:
indeed., thank you igrr

shaneaj, 2017-01-04:
This code has help me with my temperature sensor and MQTT project.  However when I print out the mac address it displays in the reserve order.

> MAC: 60:1:94:E:A1:D6         (this is what i want to see)
    MAC: D6:A1:0E:94:01:60     (this is how it displays)
<

I've tried to locate a method for reversing a string but can't locate anything that I can understand or implement. 

Can anyone suggest how I can get the last character to be first etc. etc?

I want to include the mac address in a json object so that I can identify what device is sending the data.

I'm just learning and any help or advice would be appreciated.

fxoislac, 2017-03-20:
Hi shaneaj,

I came on this thread because it seems that WiFi.macAddress() returns a reversed mac address (I think I have the same problem you have), so I had to reverse it in order to get the correct mac address to set in the "wifi protection by mac address" option on my router.

So basically, if you take the provided example by Arduino: [https://www.arduino.cc/en/Reference/WiFiMACAddress]  

You can do something like that from Humancell topic above:

`
String getMacAddress() {
	byte mac[6];
	byte mac_reversed[6];
	WiFi.macAddress(mac);
	
	for(int idx = 0; idx<6; idx++){
		mac_reversed[idx] = mac[6-idx];
	}
	
	String cMac = "";
	for (int i = 0; i < 6; ++i) {
		if (mac_reversed[i]<0x10) {cMac += "0";}
		cMac += String(mac_reversed[i],HEX);
		if(i<5)
		cMac += ""; // put : or - if you want byte delimiters
	}
	cMac.toUpperCase();
	return cMac;
}
`

devyte, 2017-03-20:
mac[6-idx] -> mac[5-idx]

Off by one.

https://www.arduino.cc/en/Reference/WiFiMACAddress]

<https://github.com/esp8266/Arduino/issues/313#issuecomment-287719906>, or mute
the thread