esp8266/Arduino #314: Flash Size setting bug

Testato, 2015-05-23:
Installed last Arduino Core by your Board Manager json url
- Choose "Generic ESP8266 Module" 
- Select "Flash Size": 4M
- Compile

Result: Sketch uses 197,372 bytes (37%) of program storage space. Maximum is 524,288 bytes.

The Flash Size setting do not work, on every value that you try, 4M, 2M, 1M, etc Arduino IDE compile always for 512K

holgerlembke, 2015-05-24:
As far as I understand the concept, this setting is currently used to adjust the size of the flash file system.

Testato, 2015-05-24:
If you try the other two official board present in this core, and compile, you receive a compilation message about 2M of flash and 4M of flash, so it is related to flash size not Spiffs size

igrr, 2015-05-24:
The concept of "Maximum program storage space" is a bit complicated, as we have two segments for code (.text and .irom0.text) which may be filled independently. This message is not easy to customize because it is generated by the IDE. The best we can do is show the space used/available in one of the segments — probably .irom.text because this is where user's code and libraries go by default.

Testato, 2015-05-24:
thanks for the explanation,
so is it necessary correct the other official boards too:
- modwifi.upload.maximum_size=2097152
- nodemcu.upload.maximum_size=4194304

a question, if i use a 4MB flash, why the maximum code space can be only 1MB and the other 3MB must be for SPIFFS ? Is it not possible use 2M+2M ?

igrr, 2015-05-24:
Thanks, I forgot to update other boards.

I haven't yet seen a sketch which would use over 1M flash. If you have such
a use case, I can investigate.

igrr, 2015-05-24:
@Testato The reason this property is important on AVRs is that they use a generic linker script which doesn't define the correct program segment size (as it varies from one part to another). With ESPs we use different linker scripts for each flash size, so linker know exactly how much code will fit. So when you try to put too much code, you will get an error from the linker which will say that "section .irom0.text will not fit into segment irom0_seg", and the IDE will not try to upload the sketch.

Testato, 2015-05-24:
And this is true also in the other way ? Example if we set maximum_size to 512KB, but we have a real bigger Flash memory, Arduino IDE do not stop the upload ?

igrr, 2015-05-24:
If you build the sketch with a linker script for 512k flash and the upload it to 1M flash, it will run.

In fact, it's not easily possible to identify the real flash size during upload... The only way I know of is to read the flash ID and hope that the second byte will give you the flash size (as in 2^n). Unfortunately this doesn't hold true for all flash chip manufacturers.

Testato, 2015-05-24:
Ok, thanks
but the update of the board.txt, like you done, remain a good think also, it is usefull for remember how many space you have.

In arduino is there another very usefull info, the static ram usage info. Is it simple implement it on this core also ? Do you prefer i open a new issue for this ?

Example:
    Sketch uses 450 bytes (1%) of program storage space. Maximum is 32,256 bytes.
    Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables.      Maximum is 2,048 bytes.

igrr, 2015-05-24:
RAM usage is also shown after the update:
```
Sketch uses 169,224 bytes (17%) of program storage space. Maximum is 983,040 bytes.
Global variables use 43,568 bytes (53%) of dynamic memory, 
leaving 38,352 bytes for local variables. Maximum is 81,920 bytes.
```

Testato, 2015-05-24:
When se van update via board manager to thus versione ?
OT: i see that the SDK arrived to 1.1, and is there interesting news Luke a new api forma RF power setting from 1to82, Hood forma reduce consumer.
Normally when you update the SDK ? Every new versione or laterali ?

closed 2015-07-30 by igrr