esp8266/Arduino #315: SDK update is not compiling

Links2004, 2015-05-24:
the commit ae15989858802848a1c0e048fd859781569d9683 is not compiling.

```
Building file: hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp
Starting C++ compile
"hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-Ihardware/esp8266com/esp8266/tools/sdk//include" -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=160000000L -DARDUINO=161 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266   -I"\hardware\esp8266com\esp8266\cores\esp8266" -I"\hardware\esp8266com\esp8266\variants\generic" -I"\hardware\esp8266com\esp8266\libraries\ESP8266WiFi" -I"\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src" -I"\libraries\OneWire" -I"\hardware\esp8266com\esp8266\libraries\Wire" -I"\hardware\esp8266com\esp8266\libraries\ESP8266WebServer" -I"\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src" -I"\hardware\esp8266com\esp8266\libraries\SPI" -I"D:\Coding\avr\arduino lib\Adafruit-GFX-Library" -I"D:\Coding\avr\arduino lib\Adafruit_ILI9341_org" -I"\libraries\SD" -I"\libraries\SD\src" -MMD -MP -MF"Libraries/ESP8266WiFi/src/ESP8266WiFi.cpp.d" -MT"Libraries/ESP8266WiFi/src/ESP8266WiFi.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp"  -o  "Libraries/ESP8266WiFi/src/ESP8266WiFi.cpp.o"   -Wall
hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp: In member function 'void ESP8266WiFiClass::beginSmartConfig()':
hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:473:76: error: invalid conversion from 'void (*)(void*)' to 'sc_callback_t {aka void (*)(sc_status, void*)}' [-fpermissive]
     smartconfig_start(SC_TYPE_ESPTOUCH, &ESP8266WiFiClass::_smartConfigDone);
                                                                            ^
In file included from hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:30:0:
hardware/esp8266com/esp8266/tools/sdk/include/smartconfig.h:25:6: error:   initializing argument 2 of 'bool smartconfig_start(sc_type, sc_callback_t, ...)' [-fpermissive]
 bool smartconfig_start(sc_type type, sc_callback_t cb, ...);
      ^
hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp: In member function 'bool ESP8266WiFiClass::smartConfigDone()':
hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:489:35: error: 'smartconfig_get_status' was not declared in this scope
     return smartconfig_get_status() == SC_STATUS_LINK_OVER;
                                   ^
hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:490:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
In file included from \hardware\esp8266com\esp8266\cores\esp8266/Arduino.h:39:0,
                 from hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h:24,
                 from hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:32,
                 from hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:22:
\hardware\esp8266com\esp8266\cores\esp8266/esp8266_peri.h: At global scope:
\hardware\esp8266com\esp8266\cores\esp8266/esp8266_peri.h:71:16: warning: 'esp8266_gpioToFn' defined but not used [-Wunused-variable]
 static uint8_t esp8266_gpioToFn[16] = {0x34, 0x18, 0x38, 0x14, 0x3C, 0x40, 0x1C, 0x20, 0x24, 0x28, 0x2C, 0x30, 0x04, 0x08, 0x0C, 0x10};
```

if i commend this lines out i get:
```
arduino.ar(Esp.cpp.o): In function `EspClass::wdtEnable(WDTO_t)':
Esp.cpp:(.text+0xfc): undefined reference to `wdt_feed'
arduino.ar(Esp.cpp.o): In function `EspClass::wdtDisable()':
Esp.cpp:(.text+0x106): undefined reference to `wdt_feed'
```

igrr, 2015-05-24:
Should be fixed in 25b9494.
Unfortunately, wdt_feed doesn't exist in this SDK version so ESP.wdtFeed() will not work now.

Links2004, 2015-05-24:
fixed.

closed 2015-05-24 by Links2004