esp8266/Arduino #316: Automated builds from appveyor fail

ozayturay, 2015-05-24:
All items in the near history are red @ https://ci.appveyor.com/project/igrr/Arduino/history so there are no downloads in artifacts. And the green ones do not work because of old java version, though it can easily be fixed with replacing java folder with a newer one from official Arduino build or Java Installation.

@igrr Please fix appveyor builds, so windows users can follow newest additions easily. :)

igrr, 2015-05-24:
Now that we have the board manager package updated every couple of days, do you think downloading the full installer is still a good option?

ozayturay, 2015-05-24:
Yes i do, unless the DTR and RTS patch make place in official Arduino IDE for the Serial Monitor to work without problems side by side with autoreset feature.

igrr, 2015-05-24:
OK will investigate what has happened to windows build.

ozayturay, 2015-05-24:
Thank you for the interest. Is there a chance for DTR and RTS patch pull request in official Arduino source?

Toshik, 2015-05-27:
https://github.com/esp8266/Arduino/pull/350 fix windows build

igrr, 2015-05-29:
Thanks to @Toshik Windows builds are back online.

Regarding the patch, Arduino.cc guys are not very active to reply to my posts on the mailing list, so i'm not sure the DTR/RTS patch will be mainlined soon.

closed 2015-05-29 by igrr

ozayturay, 2016-07-15:
It's broken again. The last succesfull compile with an artifact is https://ci.appveyor.com/project/igrr/arduino/build/0.0.153/artifacts and i'm using this build with a portable subfolder so that ESP8266 specific BM packages don't mix up my latest Arduino enviroment.

But i guess for new functions like OTA Update from Arduino IDE version 1.6.7 or newer is needed.

http://esp8266.github.io/Arduino/versions/2.3.0/doc/ota_updates/readme.html#arduino-ide

@igrr : Can you please fix it so that we can have a recent artifact at least one more time. Thank you. :)

igrr, 2016-07-17:
@ozayturay as you may have noticed, IDE files have been removed from our repository about a year ago, as was appveyor.yml build script. We no longer support building the complete IDE package.

The method you use with portable subfolder makes sense, that's what i'm doing myself to keep different versions separate.

If what you need is the DTR/RTS patch, feel free to do a git clone of https://github.com/arduino/Arduino/pull/4102 and build IDE from source.

ozayturay, 2016-07-17:
Oh i haven't noticed it since i was using board manager with 0.153 appveyor artifact. I better do as what you said and compile it myself one more time. :)

cranphin, 2016-08-25:
Is that a 1.6.9 with support for controlling DTR/RTS in serial monitor? I'd be interested :) No idea why it isn't in main yet tho.. XD

ozayturay, 2016-08-25:
Indeed. I have compiled currently latest 1.6.11 version with the same patch and it works as expected. You can download it from my https://github.com/ozayturay/Arduino-ESP8266 repository.

The package you download is only the Arduino IDE, you have to manually get the ESP8266 toolchain via board manager or any other method as instructed at https://github.com/esp8266/Arduino repository.

The archive has an empty folder named portable in it for not using user directories for the toolchain, if you don't want a portable installation just delete this folder before using the board manager.

cranphin, 2016-08-25:
Works! Awesome, thanks :)

ozayturay, 2016-08-26:
Compilation instuctions in the comment https://github.com/esp8266/Arduino/issues/22#issuecomment-91043388 updated, 1.6.11 compiled without problems and works as expected.