esp8266/Arduino #317: upload to ESP via Arduino DUE don't work

steamboating, 2015-05-24:
I tried to upload code out of the Arduino IDE via a DUE into the ESP-01 like described here:
http://www.instructables.com/id/Use-Arduino-Due-to-program-and-test-ESP8266/?lang=de&ALLSTEPS

With the Arduino IDE 1.6.4 on my Win 8.1 system I can establish a serial monitor connection via the DUE to the ESP and communicate.

I also can upload new firmware through the DUE mirroring program with the ESP8266 Flasher and use it with the LuaLoader through the DUE.

When trying to upload code with the Arduino IDE through the DUE I get the blue LED flashing as serial request but there is no upload connection:

trying to connect
    flush start
    setting serial port timeouts to 1 ms
    setting serial port timeouts to 1000 ms
    flush complete
    espcomm_send_command: sending command header
    espcomm_send_command: sending command payload
    read 0, requested 1

I get the above message several times and it ends with

warning: espcomm_sync failed
error: espcomm_open failed

I integrated a 
  Serial.setTimeout(1000); 
into my mirroring program alreeady - still no connection.

When using the FTDI Adapter from an UNO board without the UNO Chip installed I can upload programs to the ESP without problems - even with the 3.3 Volt still from the DUE. During the upload the blue LED on the ESP is flashing much faster than during the serial requests described above - quite obviouse ...

I need a way to program the ESP through the DUE because I have developed a board with an Atmel SAM3X8 (DUE) and an ESP8266 on it as toy robot platform...

Thanks for any help.
Rainer from Germany
www.steamboating.de

igrr, 2015-05-24:
Unfortunately I don't have DUE to test this. Can you capture what happens on the serial line using a logic analyzer?

steamboating, 2015-05-24:
Thanks for the quick reply!

It's funny that all other ESP programs that I have used before work with this DUE mirroring but not the Arduino IDE...

Unfortunately my logic analyzer is still in the China Post ;-)
Will try to get one from a friend and show what's going on there...

Pablotis, 2015-05-24:
Happens the same to me with MEGA and DUE

steamboating, 2015-05-24:
Hello igrr and Pablotis,
I made a quick setup with the MEGA - yes - I get the same problem as I have described above:

"serial monitor" communication through this mirroring prorgam is possible but not the programing.

[image]

@igrr 
Will still try to get the logic analyser but 
maybe you have a MEGA at hands?

Rainer from Germany

igrr, 2015-05-24:
I don't have a Mega but I do have a Leonardo somewhere, I think it has two serial ports as well (one over USB and one UART). Can you point me to the mirroring program that you use?

steamboating, 2015-05-24:
Thanks for helping us!

I used it as ddescribed here:
http://www.instructables.com/id/Use-Arduino-Due-to-program-and-test-ESP8266/?lang=de&ALLSTEPS

// =======================================================================
#define Serial_ESP Serial3
#define Serial_Debug Serial

#define Serial_Baud 115200
 
// =======================================================================
void setup() {
  Serial_ESP.begin(Serial_Baud);
  Serial_Debug.begin(Serial_Baud);
}
// =======================================================================
void loop() {
  if (Serial_ESP.available()) {
    byte c = Serial_ESP.read();
    Serial_Debug.write(c);
  }
  if (Serial_Debug.available()) {
    byte c = Serial_Debug.read();
    Serial_ESP.write(c);
  }
}
// =======================================================================

You can run it with "byte c", "int c", or "char c"

Similar example can be found on the Arduino-cc homepage:
http://www.arduino.cc/en/Tutorial/MultiSerialMega

void setup() {
  // initialize both serial ports:
  Serial.begin(9600);
  Serial1.begin(9600);
}

void loop() {
  // read from port 1, send to port 0:
  if (Serial1.available()) {
    int inByte = Serial1.read();
    Serial.write(inByte);
  }
 
  // read from port 0, send to port 1:
  if (Serial.available()) {
    int inByte = Serial.read();
    Serial1.write(inByte);
  }
}

Rainer

Pablotis, 2015-05-24:
I have made a log with an arduino due of the data transmitted during the procces.
I have done two logs, the first uploading a firmware from arduino ide, that didn't upload, and the other using the flash tool, that uploaded succesfully.
https://drive.google.com/folderview?id=0BxPwW0n9dYaFfkVFTy1tUG96QkVCQi1xSC1YdEJ2d2o3Xy1KZ09wbFpoRjdPQnBHVWhVSG8&usp=sharing

igrr, 2015-05-24:
How to interpret this log? I see a lot of '1's...

steamboating, 2015-05-24:
@igrr 
Looks for me like the only the "send line" from the ESP? The log "Arduino IDE" contains only 1 because nothing comes back . The "ESP8266 Flash Downloader Tool.log"contains some 01 sequences

@Pablotis 
Maybe you can send a log file with parallel rx and tx signals - maybe as a screenshot?

Rainer

Pablotis, 2015-05-24:
I have uploaded to the same folder 2 new logs that contains data sended from the arduino ide when programming and from the uploading tool, if you want i could make the same with the responses from the esp, the firmware i have uploaded with the tool is the official AT commands.

steamboating, 2015-05-24:
Thank you Pablotis for this second upload.
Hopefully igrr knows how to interpret this info...

Rainer

Pablotis, 2015-05-24:
I can also upload the commands of the previous version of the arduino ide, the ones that works

steamboating, 2015-05-24:
@Pablotis 
Which IDE worked for you?
Which ESP version have you used with it?
Is that still downloadable from any place?

Would love to install this for my actual trials. On my own ATMEL SAM3X8 based PCB I have wired the ESP direct to the SAM - this PCB has no FTDI driver and even no connectors for such a connection. 

Rainer

Pablotis, 2015-05-24:
I have an ESP01 which i bought on banggod, the version that works for me is the 1.6.1 of the Arduino IDE, i think you can download it on the relases page but they have removed the link so i don't know how to acces, if you want i can upload it to Google Drive but with my connection it will take 2 houres aproximately.

steamboating, 2015-05-24:
The Arduino 1.6.1 is still here:
http://www.arduino.cc/en/Main/OldSoftwareReleases#previous

But which ESP tool version have you used? And where to find that?
Rainer

Pablotis, 2015-05-24:
I meant the version of the IDE 1.6.1 that cames with esptool installed, it wasn't an official download from the Arduino web, it was from the project, maybe @igrr can give you the link.

igrr, 2015-05-24:
https://github.com/esp8266/Arduino/releases

Pablotis, 2015-05-24:
Have you seen anything strange on the log?

steamboating, 2015-05-24:
@igrr 
Thanks a lot for the link - this shows the esptool programer - but I am getting a stupid "Error compiling" even before uploading with all the ESP examples and can't test the issue with the mirroring...

Don't know why and you should better work on the issue with the 1.6.4 IDE than with this old stuff...
Meanwhile I can lern to program with the great ESP IDe content from the 1.6.4 IDE without the access over the DUE and MEGA.

Rainer

------------------------------------------
Arduino: 1.6.1 (Windows 8.1), Board: "Generic ESP8266 board"

Using library ESP8266WiFi in folder: D:\arduino-1.6.1\hardware\esp8266com\esp8266\libraries\ESP8266WiFi 

D:\arduino-1.6.1/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -ID:\arduino-1.6.1/hardware/tools/esp8266/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10601 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -ID:\arduino-1.6.1\hardware\esp8266com\esp8266\cores\esp8266 -ID:\arduino-1.6.1\hardware\esp8266com\esp8266\variants\esp01 -ID:\arduino-1.6.1\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src C:\Users\ra\AppData\Local\Temp\build752787840858990034.tmp\WiFiScan.cpp -o C:\Users\ra\AppData\Local\Temp\build752787840858990034.tmp\WiFiScan.cpp.o 

Error compiling.

Pablotis, 2015-05-27:
Any fix?

steamboating, 2015-05-27:
@igrr 
My Saleae logic analyzer arrived today - what kind of reading do you need?

Pablotis, 2015-05-30:
No response...

steamboating, 2015-05-30:
Wood be great to get back this connectivity between Arduino and ESP!

For my oppinion many people would like to use an Arduino as "programmer" for the ESP. Especially when using an Arduino for the sensors and external functions like Displays and so on and the ESP for wlan communication. Would make programming and testing much easier than using a Serial/USB adapter all the time and having the need to rewire the ESP to the Arduino afterwards all the time!

Pablotis, 2015-06-02:
=(

hariprasadiit, 2015-06-03:
I'm able to upload program to ESP using DUE with any problem.Make sure you have check the following

1.Arduino IDE baud rate and baud rate used in the DUE program should be same.
2.Make sure you use common ground if you are using external power supply for ESP.

this is the program i used.

// =======================================================================
#define Serial_ESP Serial3
#define Serial_Debug Serial

#define Serial_Baud 115200

// =======================================================================
void setup() {
Serial_ESP.begin(Serial_Baud);
Serial_Debug.begin(Serial_Baud);
}
// =======================================================================
void loop() {
while (Serial_ESP.available()) {

Serial_Debug.write(Serial_ESP.read());
}
Serial_Debug.flush();

while (Serial_Debug.available()) {

Serial_ESP.write(Serial_Debug.read());
}
Serial_ESP.flush();
}

steamboating, 2015-06-03:
Hello hariprasadiit
great news! Which Arduino IDE have you used. 

Baud rate and ground should be OK in my case because I am able to use the DUE as Serial to USB adapter with other ESP software already.

Hopefully your flush() command will do the thing. Will try it out as soon as I am back home from work.

Rainer

hariprasadiit, 2015-06-03:
I've used V1.6.4 with ESP libraries installed via Board manager

steamboating, 2015-06-03:
Thank you!
Your explanations let me hope to solve this problem easy soon. 
Will tell you about my trials this evening (German time).
Rainer

Pablotis, 2015-06-03:
It worked for me!!! a lot of thanks!!!

steamboating, 2015-06-20:
@Pablotis 
came back to this issue - it even don't work with the programm above for me - not with the DUE, not with the MEGA. Please let me know your GPIO setting and procedure for switching in prog mode.

Rainer

Links2004, 2015-11-06:
If it still a problem feel free to reopen.

closed 2015-11-06 by Links2004

Bushii7, 2017-07-26:
same problem with MEGA 
any help?

Arduino: 1.8.2 (Windows 8.1), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), nodemcu, Disabled, None"

Archiving built core (caching) in: C:\Users\Bushra\AppData\Local\Temp\arduino_cache_628947\core\core_esp8266_esp8266_generic_CpuFrequency_80,FlashFreq_40,FlashMode_dio,UploadSpeed_115200,FlashSize_512K64,ResetMethod_nodemcu,Debug_Disabled,DebugLevel_None_____cb4648a41a8648d50fd87beadf221914.a
Sketch uses 222205 bytes (51%) of program storage space. Maximum is 434160 bytes.
Global variables use 31572 bytes (38%) of dynamic memory, leaving 50348 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed