esp8266/Arduino #318: FS.check() raises wdt-reboot

holgerlembke, 2015-05-24:
Might be a problem on 4m-chips only?

FS Mount: 1
FS Size: 3125248
FS Bytes: total: 2949250, used: 0
FS Blocks: total: 381, free: 381, size: 8192
FS Pages: allocated: 0, deletedPages(): 0, pageSize(): 256
FS, check(): 
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 28780, room 16 
tail 12
chksum 0xc1
ho 0 tail 12 room 4
load 0x3ffe8000, len 1580, room 12 
tail 0
chksum 0x89
load 0x3ffe8630, len 5692, room 8 
tail 4
chksum 0x25
csum 0x25

igrr, 2015-05-24:
As I [mentioned](https://github.com/esp8266/Arduino/issues/308#issuecomment-104915576) in the other issue yesterday, don't use FS API because it has been removed.

holgerlembke, 2015-05-24:
Ah, forgotten. Is there already a replacement?

closed 2015-05-24 by holgerlembke