esp8266/Arduino #322: interrupt CHANGE acts like RISING / attachInterrupt

chaeplin, 2015-05-24:
ESP-12 / pin 12 grounded with switch.

-----------------
int pin1 = 14;
volatile int state = LOW;

void setup()
{
  Serial.begin(38400);  
  pinMode(pin1, OUTPUT);
  pinMode(12, INPUT_PULLUP);
  attachInterrupt(12, blink, CHANGE);
}

void loop()
{
  digitalWrite(pin1, state);
  //
  Serial.print("state: ");
  Serial.print(state);
  Serial.print("\tread  : ");
  Serial.println(digitalRead(12));
  delay(500);
}

void blink()
{
  state = !state;
}

-------------

result :+1: 

state: 1    read  : 1 
state: 1    read  : 1 
state: 1    read  : 0 
state: 1    read  : 0 
state: 1    read  : 0 
state: 1    read  : 0 
state: 1    read  : 0 
state: 1    read  : 0 
state: 1    read  : 0 
state: 1    read  : 0 
state: 0    read  : 1 
state: 0    read  : 1 
state: 0    read  : 1 
state: 0    read  : 1 
state: 0    read  : 1 
state: 0    read  : 0 
state: 0    read  : 0 
state: 0    read  : 0 
state: 0    read  : 0 
state: 0    read  : 0 
state: 0    read  : 0 
state: 0    read  : 0 
state: 1    read  : 1 
state: 1    read  : 1 
state: 1    read  : 1 
state: 1    read  : 1 
state: 1    read  : 1

rayshobby, 2015-05-30:
I second this issue: I was trying to get RC-Switch library to work with ESP8266 so that I can receive signals from a remote control. It didn't work. After debugging, it seems interrupt mode 'CHANGE' is not capturing all rising and falling edges, instead, it's capturing only one type of edge.

hariprasadiit, 2015-06-04:
same here.CHANGE is acting like RISING.

vshymanskyy, 2015-06-11:
Experienced same problem.

igrr, 2015-06-15:
Thanks for the reports, I've found the bug, fix coming up.

chaeplin, 2015-06-16:
Test result : +1: 
---
CHANGE
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 1    read : 0 <-----
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 0    read : 1  <-----
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 1    read : 0  <-----
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 0    read : 1  <-----
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 1    read : 0  <-----
state: 1    read : 0
state: 1    read : 0
state: 0    read : 1  <-----
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1

---
RISING

state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 0
state: 0    read : 0
state: 0    read : 0
state: 0    read : 0
state: 0    read : 0
state: 1    read : 1  <-----
state: 1    read : 1
state: 1    read : 1
state: 1    read : 1
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 0    read : 1  <-----
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 0
state: 0    read : 0
state: 0    read : 0
state: 0    read : 0
state: 0    read : 0
state: 1    read : 1  <-----
state: 1    read : 1
state: 1    read : 1
state: 1    read : 1
state: 1    read : 1
state: 1    read : 1

---
FALLING

state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 1    read : 0  <-----
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 1    read : 1
state: 1    read : 1
state: 1    read : 1
state: 1    read : 1
state: 1    read : 1
state: 1    read : 1
state: 1    read : 1
state: 0    read : 0  <-----
state: 0    read : 0
state: 0    read : 0
state: 0    read : 0
state: 0    read : 0
state: 0    read : 0
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 0    read : 1
state: 1    read : 0  <-----
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0
state: 1    read : 0

closed 2015-06-16 by chaeplin