esp8266/Arduino #323: NOT_ON_TIMER is not defined

Duality4Y, 2015-05-24:
so when running the new ide and trying to run code for a i2c display it tells me that NOT_ON_TIMER is not defined ... which is not defined in Arduino.h, which if you define it to be zero it also raises an error in which it is used to check if a pin is a timer. like:
digitalPinToTimer is not defined

so this is what i added to make it go away:
#define NOT_ON_TIMER             (0)
#define digitalPinToTimer(pin)      (NOT_ON_TIMER)

basicly it's this librarie that provided this issue:
https://bitbucket.org/fmalpartida/new-liquidcrystal/downloads

igrr, 2015-05-24:
Does the library work after this change?

Duality4Y, 2015-06-06:
yes, i did a pullrequest :)

closed 2015-06-06 by Duality4Y