esp8266/Arduino #324: Package for nightly build

sanchosk, 2015-05-24:
I'd like to help with testing, however I am not experienced with building software.
It would be great if there was a link for package manager, like
http://arduino.esp8266.com/package_esp8266com_nightly.json
with overnight builds, so I can easily replace the current package with a new one and test old code.

igrr, 2015-05-24:
Good idea. I'll add a scheduled build for that.

igrr, 2015-05-29:
http://arduino.esp8266.com/staging/package_esp8266com_index.json

probonopd, 2015-06-21:
I added http://arduino.esp8266.com/staging/package_esp8266com_index.json as an additional source to my existing sources and installed 1.6.4-835-g77d77e8 using the Boards Manager. As a result, in $HOME/Library/Arduino15/packages/esp8266/hardware/esp8266 I had both 1.6.2/ and 1.6.4-835-g77d77e8/ - this was confusing the IDE to the point that it crashed immediately when starting the IDE. Removing 1.6.2/ solved this for me. Would be nice if this could work automatically.
(Please let me know if this must be filed upstream as an Arduino IDE bug instead.)

Testato, 2015-06-25:
May you update the development board manager package ?

Testato, 2015-06-26:
i updated this evening to the last development pachage whit  your staging json link.
I confirm that in the C:\Users\win7\AppData\Roaming\arduino15\staging and packages remain the old version. So the Ide do not know which one must use.
I must manually delete the old 1.6.4-835-g77d77e8 and leave only the last 1.6.5-804-g2d340c7

igrr, 2015-07-22:
Package has been updated. If all goes well, will roll it out as stable later today.

closed 2015-07-30 by igrr