esp8266/Arduino #327: ESP8266WebServer.cpp does not parse get  parameters.

far5893, 2015-05-25:
In esp8266webserver library there is not parsing of GET parameters,
this library is only for examples or can i extend with parameters parsing ?

igrr, 2015-05-25:
I think GET parameter parsing is implemented:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src/ESP8266WebServer.h#L70

Do you have any issues using ```arg```, ```argName```, ```argCount```, ```hasArg``` APIs?

far5893, 2015-05-25:
Ok , i suppose that webserver should call fn(function associated to uri) like a c main with argc/argv
array , but it's one-client/one-query server so parameters are stored   in web server object.
All it's clear now, may be more complex example with get should be done..

marvinroger, 2015-05-25:
Well, there is already an example showing get URL params:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/examples/AdvancedWebServer/AdvancedWebServer.ino#L82

far5893, 2015-05-25:
Put arg example in 404 error handler is not so friendly!!
Ok, it's' open source and you must read all code , but there is a lot of project to read code outside.......

marvinroger, 2015-05-25:
Arduino on ESP8266 is still in early stage and as you can see there is currently no real docs available. I guess you will have to dig into the code for now. But I am pretty sure it will come one day, but there is probably a lot more important stuff to do right now.

far5893, 2015-05-26:
Ok,thank.

closed 2015-05-26 by far5893