esp8266/Arduino #329: Can't install ESP8266 Boards through Board Manager in Arduino IDE 1.6.4

rvbcrs, 2015-05-25:
I downloaded the Arduino 1.6.4 IDE and copied this URL: http://arduino.esp8266.com/package_esp8266com_index.json in the additional board managers url. When I open the board manager I see the ESP8266 Package but when I press install I get the error as shown in image!

[image]

WoodsterDK, 2015-05-26:
Hi,
I had that as well - if you manually copy the file into the following folder .."Roaming\Arduino15\staging\packages" and restart the installation then the installation succeeds....

rvbcrs, 2015-05-26:
Ok, it seems to be working now! Haven't done what you suggested maybe Ivan fixed it :)

igrr, 2015-05-26:
I haven't touched anything on the server today, so might have been an intermittent issue...

closed 2015-05-26 by rvbcrs

faisalzed, 2017-06-02:
you mean i need to change the link to the preferences  ?

harshgar, 2018-01-20:
Its probably just a network error keep trying it will work.