esp8266/Arduino #331: UDP stream causes hang

sticilface, 2015-05-25:
I've managed to get UDP streaming to work with the ESP and WS2812s using pixelcontroller (http://pixelinvaders.ch/?page_id=160)  looks fantastic... 

The problem is that is seems to cause it to hang after a while.  Im not sure why, it only happens when streaming UDP to it...  here is the code... 

Is there anyway of working out why this is.. or putting some safeguards inlace to say drop packets if the buffer is running out?  This is my first use of UDP streaming ever... 

here is the code...

```

void UDPfunc () {
int packetSize = Udp.parsePacket();

  if(Udp.available())
  {
    RgbColor col;
    int currentpixel = 0;
    for (int i = 0; i < packetSize; i = i + 3) {
      if (currentpixel > pixelCount) break;
        col.R = Udp.read();
        col.G = Udp.read();
        col.B = Udp.read();
        strip->SetPixelColor(currentpixel,col);
        currentpixel++;
      }
strip->Show();
}
```

sticilface, 2015-05-26:
A little more information.  It seems to just hang the web server part of it.  The ESP is still processing serial input, and in my case streaming it to some WS2812s... The web server just stops responding.

Links2004, 2015-11-06:
please retry in last release, If it still a problem feel free to reopen.

closed 2015-11-06 by Links2004