esp8266/Arduino #335: No user libraries with Visual Micro

WoodsterDK, 2015-05-26:
Hi there,

I'm using the Visual Micro plugin, and have installed the ESP8266 toolchain from the board manager.
I'm missing the ESP8266 plugins in the project, and cannot include them.
Is there a quick fix for adding them - that still takes the board manager approach in account?
Maritn

RussMathis, 2015-05-26:
For Visual Studio 2013; With Arduino 1.6.4, set "Preferences/Show verbose output during compilation" checked, and build and esp sketch. Then review the verbose output, there you'll find where all the files being included are. Next build and ESP sketch, again with verbose on from "Tools/Options/VisualMicro/verbose" set to true.

I started by downloading the "ESP Arduino IDE" and unpacking to my "Sketch/hardware" directory. Then pointing the "Tools/Options/Visual Micros/Application Ide Locations" for the ESP8266 board to my Arduino-1.6.4 directory, then the sketch directory to my sketch directory with the hardware folder where the "ESP Arduino IDE" was unpacked.

Next by comparing outputs from the original Arduino 1.6.4 build I was able to find the missing folders that MicroVisual was looking for and copied them to the sketch/hardware folder.

Also noticed when testing build from Visual Studio I had to change the board type from ESP8266 to an Arduino board choice and back to the ESP8266 before building. Something is not correctly be set when leaving the "tools/options" menus. Even if I tried the "Tools/VisualMicro/Rebuild Tool Chain" the switch from ESP8266 board to Arduino and back was required before builds would successfully work.

Good look.

WoodsterDK, 2015-05-27:
Hi there,
now it is working. I simply selected the arduino board type, and then the ESP8266. I guess it has issues reloading the tool chain correctly.

closed 2015-05-27 by WoodsterDK