esp8266/Arduino #337: Question : what is correct SDK  version in 1.6.4-673-g8cd3697?

luc-github, 2015-05-26:
Hi I use the board manager which use 1.6.4-673-g8cd3697 package 
Changelog (https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/changes.md) says SK is : 1.1.0

but if I use system_get_sdk_version() function, I get : 1.0.1

Is it a typo issue ? which one is correct ?

Thanks for your great work

igrr, 2015-05-26:
Well, since you are using 164-673-g8cd3697, you need to refer to the corresponding section of the change log:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/changes.md#164-673-g8cd3697

igrr, 2015-05-26:
Related to #333.

luc-github, 2015-05-26:
Ok my bad - I assumed wrongly the 1.6.4-673-g8cd3697 was current version, I do no see well the horizontal lines on my screen ... I am getting old

Thanks for the quick clarification

closed 2015-05-26 by luc-github