esp8266/Arduino #338: Using ADC

fchanson, 2015-05-26:
Hi,

I made some tests on the ADC using a NodeMCU 0.9.5 build 20150318.
My code just reads the value on the pin every 500ms.

Using a Arduino program : 
- pin A0 at GND --> analogRead(A0) = 700
- pin A0 at 3.3V --> analogRead(A0) = 909

I am surprised to have the value 700 when voltage on pin A0 is GND. Should be near 0.

I made the same test with a LUA program using ESPlorer.
The result is the following :
- pin A0 at GND --> adc.read(0) = 4
- pin A0 at 3.3V --> adc.read(0) = 1023

These last results sound good.

Does anyone have the same problem ?

Need something special to use the ADC pin ?

Thanks.

fchanson, 2015-05-27:
Hi,

I found in the file 'core_esp8266_wiring_analog.c' the code used for the analogRead method.

extern int __analogRead(uint8_t pin) {
  if(pin == 17){
    return readvdd33() >> 2; // readvdd33 is 12 bit
  }
  return digitalRead(pin) * 1023;
}

Does anyone can explain me this part of code ?
Why the method returns the VDD value divided by 4 ?

Thanks a lot.

igrr, 2015-05-27:
This seems to be related to having or not having esp_init_default_data.bin flashed.
Apparently if you have it (for instance from a previous NodeMcu install), readvdd33 returns the voltage on VDD pin. If you don't, readvdd33 reads the value from the TOUT pin.

fchanson, 2015-05-27:
Thanks for your reponse.

I don't know how it is possible to "remove" this file during the flash step.

Could you please help me on this point ?

bluewalk, 2015-05-27:
Having this exact same issue, is there anything we can do to work around this?

czopas, 2015-05-29:
Hello, I have the same issue like friends before. I am using voltage divider 10k / 33k so on TOUT (ADC pin) I have 850mV. The result of analogRead(A0) is about 20. I have flashed ESP12 with firmware v0.9.5.2 AT Firmware.bin.
As I think, I should reflash my ESP12 with esp_init_default_data.bin and then ADC should work correct, right ? Where I can find this file ?

bluewalk, 2015-06-02:
`As I think, I should reflash my ESP12 with esp_init_default_data.bin and then ADC should work correct, right ?`
I've just tried that, according to this file https://developer.mbed.org/media/uploads/sschocke/esp_flasher.zip (which includes esp_init_default_data.bin) you need to flash it at address 0x7C000. This just breaks the whole arduino code and esp won't boot until you reupload your sketch. Which, then again, has the same not working ADC bug.

igrr, 2015-06-02:
This esp_flasher has a bug which causes the first few sectors of flash to
be erased when you program the sector at 0x7c000. This is why the sketch
will not boot after flashing init_default.bin.
Regarding how this affects ADC, well, too bad it's not fixed by flashing
this config.

bluewalk, 2015-06-02:
Yes, too bad indeed. Do you have any other idea's that might help us finding a solution?

igrr, 2015-06-02:
Sure.
Can you try reverting [this change](https://github.com/esp8266/Arduino/commit/09bb75874deb95c22b4c09e63b842fc1f89eebc3) to see if that fixes ADC? On the module i have here both versions work reasonably well, but apparently not so on your side.

bluewalk, 2015-06-02:
Reverting the change provided fixes the issue I have with my boards!

igrr, 2015-06-02:
Do you have any external resistors connected to the TOUT pin? Resistor divider?

bluewalk, 2015-06-02:
Yes, I'm using the following schematic: 
[image]
It's part of the NodeMCU dev board. I'm also using this schematic in one of my own designed pcb's.

czopas, 2015-06-02:
Great news :)

Where should I change core_esp8266_wiring_analog.c in my arduino folder ?

bluewalk, 2015-06-02:
You should change this file: C:\Users\<<USERNAME>>\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\cores\esp8266\core_esp8266_wiring_analog.c

igrr, 2015-06-02:
So we have two groups of users, and two versions. One version works for one group, another works for the rest. Have to figure out what the difference is.

czopas, 2015-06-02:
It's no working for me :( When I pull down ADC pin to ground and use 
Serial.println(analogRead(A0)); 
I received values about 440 ...

Where is a problem ?

bluewalk, 2015-06-02:
I've just tried a ESP-12 besides my NodeMCU dev board and with ADC grounded I get readings around 380-400 and sometimes a 0.
When using the setup as shown in the schematic above, most readings are 0, but it has spikes to around 430. Applying 3.3V to the ADC pin using the above schematic gives me again most of the times 1023 but also readings around 430.

It seems that on an ESP that is not integrated on a dev-board the readings are sometimes off.

Grounded:
```449
450
0
419
0
0
0
0
416
416
```

To VCC:
```480
480
1023
1023
482
481
481
481
1023
548
1023
```

fchanson, 2015-06-03:
I made the changes in the file core_esp8266_wiring_analog.c and the issue is fixed on my NodeMCU 0.9.5 build 20150318. 
I made some tests on several NodeMCU (same firmware version) and the results are consistent and stable.

To ground:
NodeMCU 1 : 2
NodeMCU 2 : 5

To VCC:
NodeMCU1 : 1023
NodeMCU 2 : 1023

to VCC/2 (resistor divider):
NodeMCU1 : 534
NodeMCU2 : 539

czopas, 2015-06-03:
Ok I will try this on NodeMCU firmware (not v0.9.5.2 AT Firmware.bin) but I remember I had a problem with programming from Arduino IDE when I flashed NodeMCU firmware :( There was a problem with communication.

fchanson, 2015-06-03:
I used the NodeMCU flasher (https://github.com/nodemcu/nodemcu-flasher) with the internal binary file (INTERNAL://NODEMCU). This config uses nodemcu_integer_0.9.5_20150318.bin.

bluewalk, 2015-06-03:
Doesn't the Arduino IDE overwrite parts of the firmware anyway? Address 0x0 and 0x1? So if I understand correctly, you've flashed NodeMCU first and then flashed your generated bins from Arduino IDE through the upload button itself and that makes a difference?

fchanson, 2015-06-03:
yes you are right.
First I flashed NodeMCU.
Next I uploaded my sketch from Arduino IDE.

bluewalk, 2015-06-03:
I'll give that a try later when I get home. See if I can replicate it.

czopas, 2015-06-03:
On what speed you upload your sketch from Arduino IDE after flashing NodeMCU ? I have "error: espcomm_open failed" :(

fchanson, 2015-06-03:
Upload Speed : 115200
CPU Frequency : 80Mhz

bluewalk, 2015-06-03:
It must be my ESP-12 that is broken somehow, tried an ESP-07 with the above fix and that one works like a charm! Or is it because it had the AI-thinker firmware on at first? Will try to flash the ESP-12 with AI-thinker firmware and report back.

Testato, 2015-06-06:
i have same rpoblem on nodemcu dev board whit the original resistor divider.

Is there also another problem, after a 3 or 4 analog reading the chip reboot and on serial i receive this message.

----------------------------
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 28780, room 16 
tail 12
chksum 0x2d
ho 0 tail 12 room 4
load 0x3ffe8000, len 1332, room 12 
tail 8
chksum 0xc6
load 0x3ffe8540, len 1576, room 0 
tail 8
chksum 0xa4
csum 0xa4
------------------------------------

czopas, 2015-06-09:
Any new solution for this problem ? I still have various values from ADC even if I flash ESP_8266_BIN0.92.bin before uploading sketch.

Testato, 2015-06-09:
Arduino Ide completely rewrite the flash, so there is no difference if you upload a .bin before Arduino upload.

czopas, 2015-06-09:
Yes I know, but someone wrote this resolve problem. I have no idea how to fix my issue :(

bluewalk, 2015-06-09:
```Arduino Ide completely rewrite the flash, so there is no difference if you upload a .bin before Arduino upload```
It doesn't rewrite the complete flash, that's the thing. Probably depending on the size if your sketch and the flash space used for EEPROM the results may vary.

What I did to resolve is revert the change provided by @igrr, flashed the ESP_8266_BIN0.92.bin and then uploaded the sketch. Note that I'm using the same resistor divider setup as I posted earlier (also the same as used on the NodeMCU devboard).

Testato, 2015-06-09:
If this is true this is a dangerous bug. On all chip, example the Atmel present on Arduino board, before upload the sketch the flash is erased by avrdude command.
This must be the same here also.
I hope that @igrr agree on this philosopy.

czopas, 2015-06-09:
I reverted the change provided by @igrr, flashed ESP_8266_BIN0.92.bin and after that upload the new sketch and still have a problem.

Testato, 2015-06-09:
thanks for the test,
This confirms what we thought

bluewalk, 2015-06-09:
I've just updated my sketch where I added a ```WiFi.mode(WIFI_STA);```. Uploaded it and suddenly saw invalid readings again, a couple of good ones and suddenly back to the 440-460 range, and then suddenly a couple of good ones and so on. Since the wifi thing was the only change, I reverted the code and reuploaded my sketch, hoping it would all be fine. Unfortunately it wasn't...

duncan-a, 2015-06-09:
Could this be a hardware issue - have you tried another board?

Just a thought...

drmpf, 2015-06-09:
My tests also show flashing does NOT completely overwrite the memory. 
 If I have EEPROM.begin(512) in my sketch, re-flashing does NOT clear the settings I have made in EEPROM.

Can some one come up with a 'Clear Memory' option/sketch for the IDE  that really completely clears the memory?

Testato, 2015-06-10:
Wait, this is normale.
The EEPROM , emulated on this chip, must retain is data after a flash.
You must manually reset it by a function that pur FF on alla EEPROM bit.

This is a differenti think respect the complete flash deleting after an upload.

bluewalk, 2015-06-10:
> Could this be a hardware issue - have you tried another board?
I would find that weird as the same board worked perfectly before I re-uploaded my little edit in the sketch.

AndyKirby, 2015-06-10:
I believe the nodemcu folk had problems getting steady values out fo the
ADC.

Their fix in the end was to disable the wifi, do an adc read then enable
the wifi. Apparently it was something to do with the wifi. Maybe
interference or fluctuating power rails, dunno the detail. I recall the
issue from lurking on their mailing list like I do here.

Might be worth trying anyway......

Cheers

Kirbs

bluewalk, 2015-06-12:
Not really a suitable solution as I, for example, need my wifi connection for a connection to my MQTT broker. Also on NodeMCU it doesn't disable wifi when reading ADC. Readings from my nodeMCU board are correct, ones from my ESP07 were good until I updated my sketch to disable the AP functionality. Will try again with enabling AP functionality and try again.

bluewalk, 2015-06-15:
Okay found a 100% reproducible relation between wrong ADC readings and the WiFi setup, as AndyKirby hinted.
When running WiFi in WIFI_AP_STA the readings are perfect, but when running WiFi in WIFI_STA the readings are sometimes okay, but most of the time totally off. 

Even though the quick fix would be running in AP_STA, this (at least for me) is not a suitable solution as I don't want to have each ESP to have a hotspot running. Maybe this pushes us into the direction of a permanent solution?

Testato, 2015-06-15:
the complete framework for the ADC measurement must be revised.
Is there two function from the original Espressif API, one for read the VDD and one for ADC.
Measuring the VDD must be done whit TOUT pin floating, and the result is only the VDD power supply measurement.
Instead is there another function for reading the ADC.
Beetwen this two possibilities it is necessary set the vdd_cost value inside a default.bin file.

@igrr actually you correctly manage the vdd_cost inside default.bin file ?
Where is this part ? I'm very interested on it and i can prepare a PullRequest for AnalogRead(A0) plus a new function PowerSupplyRead()
.

bluewalk, 2015-06-15:
Okay, I think I found a fix that seems to be working, basically the ```adc_sample``` code they use in NodeMCU here: https://github.com/nodemcu/nodemcu-firmware/blob/fc8fa2cc915aea3687b068eb0355bb84b2fc113d/app/modules/adc.c

My ```__analogRead``` in ```core_esp8266_wiring_analog.c``` looks like this now:
```
extern int __analogRead(uint8_t pin) {
  if(pin == 17){
    return 0xFFFF & system_adc_read();
  }
  return digitalRead(pin) * 1023;
}
```

tzapu, 2015-06-16:
i ve used the above snippet on a nodemcu board and with a photocell between a0 and vcc 3.3 and seems to work alright, managed to get values from 5 to 1009 with readAnalog
```
extern int __analogRead(uint8_t pin) {
  if(pin == 17){
    return 0xFFFF & system_adc_read();
  }
  return digitalRead(pin) * 1023;
}
```

igrr, 2015-06-16:
Does the behaviour of system_adc_read depend on WiFi mode (AP/STA/AP+STA) or the presence of esp_init_default_data.bin?

tzapu, 2015-06-16:
unfortunately as a previous test i flashed the nodemcu firmware then overwrote with arduino ide so i don t know if esp init default data is still there.
i ve tried now without wifi started and just with   WiFi.mode(WIFI_AP);
and seems to work fine
i hope this helps

bmccarthy321, 2015-06-17:
I'm no genius at this stuff, but the change to "return 0xFFF & system_adc_read();" is working awesome for me.  Doing my work in the arduino ide with a generic 8266 from this repository.  I was getting 50% of the time erroneous numbers around 400ish, when I was expecting 50s.  Upgraded to 1.6.4-673-g8cd3697 board manager yesterday, and then it started doing a wdt timeout every time it performed an analogRead().  After changing the wiring_analog.c it is working perfectly.  I'm using WIFI_STA.

Testato, 2015-06-18:
the official api document:

FOR ANALOGREAD:
system_adc_read
Function:
Measure the input voltage of TOUT pin 6, unit：1/1024 V
system_adc_read is only available when wire TOUT pin to external
circuitry，Input Voltage Range restricted to 0 ~ 1.0V.
• The 107th byte in esp_init_data_default.bin(0〜～127byte) is named as
“vdd33_const”, and when wire TOUT pin to external circuitry, the
vdd33_const must be set as real power voltage of VDD3P3 pin 3 and 4.
• The range of operating voltage of ESP8266 is 1.8V~3.6V，the unit of
vdd33_const is 0.1V，so effective value range of vdd33_const is [18,
36].

FOR POWER SUPPLY READ:
system_get_vdd33
Function:
Measure the power voltage of VDD3P3 pin 3 and 4, unit：1/1024 V
Note:
• system_get_vdd33 can only be called when TOUT pin is suspended
• The 107th byte in esp_init_data_default.bin（0〜～127byte）is named as
“vdd33_const“ , when TOUT pin is suspended vdd33_const must be set as
0xFF， that is 255

So i think it is good have two esp_init_data_default.bin, one will be uploaded when the user need AnalogRead, and one when the people need PowerSupplyRead
For now we can use only two fixed value 33(for analogread) 255(for powersupplyread) inside esp_init_data_default.bin

Testato, 2015-06-18:
I modified the esp_init_data_default.bin file, 
in this .zip is there one whit 33 value (for analogread) and one whit 255 value (for powersupplyread)

The original file have a 0, non an 0xFF inside, so it is never good

http://forum.arduino.cc/index.php?action=dlattach;topic=265682.0;attach=129826

czopas, 2015-06-18:
Can you make small instruction how to upload this esp_init_data_default.bin ? From what address should I upload this file ?

Testato, 2015-06-18:
simply replace the original file in SDK (directory bin)

igrr, 2015-06-18:
If the context here is the Arduino environment, placing that file in the SDK directory will not make Arduino upload esp_ini_data_default.bin to the chip. This is not currently handled in this core...

Testato, 2015-06-18:
ok thanks, so we must manually upload the file

i think this is an important feature to add to this core, because this byte is used also on the new function that adjust the RF power reading powersupply value on this pin, so actually, whit the original esp_ini_data_default.bin, that have Zero on 107th Byte, this function do not work.

From Documentation:
system_phy_set_tpw_via_vdd33
Function:
Adjust RF TX Power according to VDD33, unit : 1/1024 V
Note:
When TOUT pin is suspended, VDD33 can be got by system_get_vdd33;

Testato, 2015-06-18:
ok, i tested many times today, 
- if i use the offical system_adc_read() the ADC value is correct.
- if I use the offical system_get_vdd33 the value is always 65535

so i Think that from factory we have the 33value in the init .bin file that it is burned.

So for now untill in this core there is no init bin file uploading we can use only the analog read.
I modified the core_esp8266_wiring_analog.c (there is not necessary return the FFFF) and prepared a pullup request 
https://github.com/esp8266/Arduino/pull/443

igrr, 2015-06-25:
5b00eba adds a way to select ADC mode for the sketch — either TOUT or VDD.

pRoFlT, 2015-07-08:
Thanks bluewalk i modified that file and it is working for me now. We shall see what other pitfalls it will have once i start the wifi communication part of my code.

genguskahn, 2015-07-28:
I have been assembling an environment monitor within the Arduino IDE to email various parameters and used the LDR on the ADC(TOUT) on the ESP8266-201, this resulted in the device regularly crashing....after some tinkering with the circuit with no success I used the advice above indicated by Testato  and made no other changes and this has been stable for more than 24 hours... 

•if i use the offical "system_adc_read()" the ADC value is correct.

                                  /-----[LDR]---GND
I use Vcc<--[22K]---<
                                   \----[10K]-----GND

This is the output without the MQ or PIR they are triggers for the email....

Logging Temperature: 19.30 deg Celsius - Humidity: 56.40% - Time: 10:55:31 - 28/7/2015 - Distance = 1cm - Light Value = 99% Saturation
Logging Temperature: 19.30 deg Celsius - Humidity: 56.30% - Time: 10:56:35 - 28/7/2015 - Distance = 1cm - Light Value = 6% Saturation
Logging Temperature: 19.30 deg Celsius - Humidity: 56.50% - Time: 10:57:39 - 28/7/2015 - Distance = 1cm - Light Value = 14% Saturation

The ADC Line in the Arduino IDE...

int LightV = map((system_adc_read()), 40, 900, 0, 100) ;
  Serial.print(100-LightV);

Hope this is of help.......

igrr, 2015-07-28:
@genguskahn are you using the latest stable build? ADC should work there without further changes.

genguskahn, 2015-07-28:
Arduino IDE 1.64 with the board manager addition only.......
But this is now very stable........and accurate.....

[13 more replies]

closed 2015-07-28 by igrr