esp8266/Arduino #341: sprintf(,"%f",); does not work, gives back %f

holgerlembke, 2015-05-26:
void setup() {
  Serial.begin(115200);
  delay(5000);
  
  float dog = 43.34567;
  
  char *cat = new char[100] ;
  sprintf(cat,"%f",dog);

  Serial.println(cat);
}

void loop() {
  delay(100);

}

Instead of a number I get "%f" back...

chadouming, 2015-05-26:
This is a well known issue among arduino user

holgerlembke, 2015-05-26:
And so we don't want to fix it?

chadouming, 2015-05-26:
/* float to string
 * f is the float to turn into a string
 * p is the precision (number of decimals)
 * return a string representation of the float.
 */
char *f2s(float f, int p){
  char * pBuff;                         // use to remember which part of the buffer to use for dtostrf
  const int iSize = 10;                 // number of bufffers, one for each float before wrapping around
  static char sBuff[iSize][20];         // space for 20 characters including NULL terminator for each float
  static int iCount = 0;                // keep a tab of next place in sBuff to use
  pBuff = sBuff[iCount];                // use this buffer
  if(iCount >= iSize -1){               // check for wrap
    iCount = 0;                         // if wrapping start again and reset
  }
  else{
    iCount++;                           // advance the counter
  }
  return dtostrf(f, 0, p, pBuff);       // call the library function
}

you can use this to convert into a string.

chadouming, 2015-05-26:
you can use this function in the meantime

http://pastebin.com/FdBFk3F4

holgerlembke, 2015-05-27:
Thanks, but coding some float to string is not the problem.

The problem is that every noob uses the printf/sprint with floats and wastes time to find the root of the problems, not really assuming that printf doesn't work with floats...

igrr, 2015-05-27:
I think *printf style functions are not really Arduino way of doing things.
String class know how to handle conversion from floats, and this is what I
see used in most example sketches, at least those supplied with Arduino.

Having floats supported in printf would be nice for some use cases though.
If you feel like rolling such an implementation, replacing printfs we have
in ROM, please go ahead. Such contribution would be appreciated.

holgerlembke, 2015-05-27:
I think you are right with the String class way to go.

My "complain" is more about the lack of documentation and finding errors again and again that are well known. To others.

closed 2015-05-27 by holgerlembke

ricardoquesada, 2016-09-21:
Probably due to ignorance, but I prefer to use a big `snprintf` than multiple `String()`s since my intuition tells me that `String` might take more resources and might be slower (just a guess).
If I have time I'll send a patch to support `%f`

murilobarix, 2017-10-08:
printf with %f  not working with 2.2.0. That is fixed in beta?

sidewaiise, 2017-12-31:
Any progress on this issue? Was trying to use a `snprintf()` for debugging but `%<l>.<p>f` , `%f` and `%F` all seem not to work.