esp8266/Arduino #342: Memory leak when server don't reply or connection false

anhhuy0501, 2015-05-27:
Hi all,
I use this code
//**************************//
void loop() {
  delay(1000);
  connectToWebSite();
}

void connectToWebSite(){
  WiFiClient client;
 
  // Use WiFiClient class to create TCP connections
  const int httpPort = 8080;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }
  
  // We now create a URI for the request
  String url = "/plugin/postwithheader";
  
  // This will send the request to the server
  String str = String("POST ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Content-Type: application/json" + "\r\n" + 
                 "Content-Length: 14" + "\r\n" +
                 "api_key: thinktodo" + "\r\n" +
                 "Connection: close\r\n\r\n"+
                 "{\"code\":\"abc\"}" + "\r\n" +
                 "";
  client.print(str);
  delay(10);
  
  // Read all the lines of the reply from server and print them to Serial
	String line="No reply....";
  while(client.available()){
    line = client.readStringUntil('\r');
  }

  num_send++;
  Serial.println(line);
  Serial.print("Free heap memory: ");
  Serial.println(ESP.getFreeHeap());
  Serial.print("Send times: ");
  Serial.print(num_send);
  Serial.println("*********");

  client.flush();
  client.stop();
}
//*******************************//

The result is:

<code>
{"status":1,"data":{"_id":"555c155c7834b3bf33ccd43e","__v":0,"status":true,"code":"abc"}}
Free heap memory: 31928
Send times: 1*********

No reply....
Free heap memory: 32160
Send times: 2*********

No reply....
Free heap memory: 32192
Send times: 3*********

connection failed

{"status":1,"data":{"_id":"555c155c7834b3bf33ccd43e","__v":0,"status":true,"code":"abc"}}
Free heap memory: 32144
Send times: 4*********

{"status":1,"data":{"_id":"555c155c7834b3bf33ccd43e","__v":0,"status":true,"code":"abc"}}
Free heap memory: 31896
Send times: 5*********

No reply....
Free heap memory: 31824
Send times: 6*********

connection failed

{"status":1,"data":{"_id":"555c155c7834b3bf33ccd43e","__v":0,"status":true,"code":"abc"}}
Free heap memory: 31728
Send times: 7*********

{"status":1,"data":{"_id":"555c155c7834b3bf33ccd43e","__v":0,"status":true,"code":"abc"}}
Free heap memory: 31728
Send times: 8*********

{"status":1,"data":{"_id":"555c155c7834b3bf33ccd43e","__v":0,"status":true,"code":"abc"}}
Free heap memory: 31728
Send times: 9*********

{"status":1,"data":{"_id":"555c155c7834b3bf33ccd43e","__v":0,"status":true,"code":"abc"}}
Free heap memory: 31728
Send times: 10*********
</code>

The free heap memory will reduce to small number and ESP8266 wdt reset. 
Any help will be much appreciated.
Thanks in advance!

cimba007, 2015-07-03:
Hi anhhuey,

I experience similar problems .. maybe they are related, maybe not .. 

I use the following code http://pastebin.com/LNxWBUWv
 with client.stop() at the end and get the following results...
```
...
i: 0 | FreeHeap before worker2()20960 | fail (worker)<\r><\n>
...
i: 99 | FreeHeap before worker2()17528 | 29.75<\r><\n>
```
 .. after 100 iterations 4k heap lost ..

using stopall() yields better results for me
```
...
i: 0 | FreeHeap before worker2()20992 | 29.56<\r><\n>
...
i: 99 | FreeHeap before worker2()20768 | 29.56<\r><\n>
...
```

I came to this issue after I noticed sometimes I was not getting a reply from my server so I investigated into multiple retrys and came to your issue.

anhhuy0501, 2015-07-04:
Hi cimba, 

Thank for your reply. I would try your way. However the issue depend on the stable of server, which is random so I don't know how to repeat it exactly in every try. 
I also try some another way and found that if you replace the esp8266-01 by the esp8266-07, also give better result (but not eliminated the issue).

cimba007, 2015-07-04:
In my case my server is 100% stable. As using a HTTP-Sniffer like "Fiddler" shows me, that connections from my windows mashine take only a few milliseconds.

One way to deal with bad connections for me is using this part of the script:

```
`    int c = 0;
    while((client2.available()<600) && c++ < 30) {
      delay(20);
      /*
      if(c == 1)
        Serial << "no data yet (odroid/graphite) ";
      else
        Serial << ".";
      */
    }
```

aka wait 600 ms or until you received 600 bytes. If this condition is not fullfilled I return "false" from my connect function and try the same again on a high lvl
```
`    int c = 0;
    while((client2.available()<600) && c++ < 3) {
      delay(20);

      "Connection function"

    }
```

Links2004, 2015-11-06:
please retry in last release, If it still a problem feel free to reopen.

closed 2015-11-06 by Links2004