esp8266/Arduino #343: PROGMEM for strings missing strncpy_P to access them

luc-github, 2015-05-27:
Hi I store my strings using PROGMEM to save some memory :
const char mystring[] PROGMEM = "my string!";

but when I want to access it to use it using strcpy_P
I have compilation error about missing strncpy_P
So I have added  manually in my project :
```
char* ICACHE_FLASH_ATTR strncpy_P(char* dest, const char* src, size_t size) {
    const char* read = src;
    char* write = dest;
    char ch = '.';
    while (size > 0 && ch != '\0')
    {
        ch = pgm_read_byte(read++);
        *write++ = ch;
        size--;
    } 

    return dest;
}
```

Is that normal strncpy_P  is missing ? or is there another way to read strings stored by PROGMEM?
If it is missing, is there any plan to add it ?
I have checked open/closed issues but nothing pop up sorry 

Thanks

Makuna, 2015-05-28:
In PgmSpace.cpp it already contains strncpy_P() so there should be no need for you to add it.  There is something else wrong.

luc-github, 2015-05-28:
here the test sketch :
```
#include <arduino.h>
#include <PgmSpace.h>
const char mystring[] PROGMEM = "this is my string";
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
}
void loop() {
  // put your main code here, to run repeatedly:
char mybuffer[50] ;
strcpy_P(mybuffer,mystring);
Serial.println(mybuffer);
delay(500);
}
```
and compilation error:
```

C:\Users\user\AppData\Local\Temp\build5883592928906091546.tmp\testme.cpp.o: In function `setup':
testme.cpp:(.text+0x28): undefined reference to `strncpy_P'
testme.cpp:(.text+0x3b): undefined reference to `strncpy_P'
collect2.exe: error: ld returned 1 exit status
Error compiling.
```

I am using the boardmanager version of ESP8266 package

Makuna, 2015-05-28:
There seems to be header conflict.
If you remove the include for Arduino.h it will compile.
If you move the include for pgmspace.h above arduino.h, it will compile.

luc-github, 2015-05-28:
actually before I did not use #include <PgmSpace.h>, just set #include <arduino.h>
I just added to see if it solved the problem but not

I did not tried change header position - just did and yes it solve the problem
Thanks a lot

But is that a normal behaviour ?

luc-github, 2015-05-28:
I guess not as if I use only ``` #include <arduino.h>``` I have compile error ```undefined reference to `strncpy_P'```

luc-github, 2015-05-28:
I changed issue title to better match problem

Makuna, 2015-05-28:
The issue is that extern "C" is incorrectly being applied inside Arduino.h to things that don't need it; in this case specifically the include for pgmspace.h as it is not a legacy "C" header file.

Links2004, 2015-05-28:
fixed https://github.com/Links2004/Arduino/commit/936669e234e997b1d0406af004d1ac56b900d4fb

luc-github, 2015-05-29:
Thanks a lot ^_^

closed 2015-05-29 by luc-github

ughodke, 2015-10-27:
FYI, downloaded and installed the EtherCard library from GitHub, and loaded up one of the sample app (SSDP).  

When using Arduino IDE version 1.6.5 with Arduino Due board, compilation fails with the same "...strncpy_P..." error. Changing the order of the include files (suggested above) did not fix the problem. So switched to other boards like Leonardo, Uno, Ethernet etc, and the compilation was successful without any errors.  

So I guess the error has something to do with board specific declarations, for the Due board or could be specific to this version of the Arduino IDE.  I hope this helps someone having a similar issue.