esp8266/Arduino #344: Using GPIO on ESP-12 Ai-Thinker module

nccchirag, 2015-05-27:
pinMode(5, OUTPUT);
digitalWrite(5, 1);

I'm not able to use any gpio ( ex 0,2,15 ).
Kindly can you please guide how to use various gpio pins alongwith the pin map.

eg. pinMode(6, FUNCTION_3); then how do we set its direction ?

igrr, 2015-05-27:
```C++
pinMode(2, OUTPUT);
digitalWrite(2, HIGH);
delay(1000);
digitalWrite(2, LOW);
```
This should toggle GPIO 2.

GPIO6 is connected to the flash chip so it would be best not to reconfigure it.
Be aware that most ESP-12 modules gave GPIO4 and GPIO5 mixed up on the silk screen...

nccchirag, 2015-05-27:
Yes 4 & 5 have been mixed up ! Thank you for the quick response
Got 4 & 5 working

closed 2015-05-27 by nccchirag

losaum, 2015-08-05:
Hi, 
Which board did you use to programming the AI-THINKER?

thanks!

nccchirag, 2015-08-06:
Used a USB-UART module - Silicon Labs CP210x USB-UART Bridge

[image]

losaum, 2015-08-06:
Hi!
But in IDE Arduino, NodeMCU 1.o or Generic ESP ... ?
For programming, you only put the jumper ?

Thanks!

nccchirag, 2015-08-06:
Generic ESP
Yes just the jumper

lucascastorina, 2016-07-19:
Hello from SPAIN!!!

IS possible to use GPIO6, GPIO7, GPIO11, GPIO8 on ESP8266-12E DEV BOARD?
I tried a simple sketch to do this, but nothing happened.
If you change to others GPIOxx seem to work well.

Thanks!!!!

lrmoreno007, 2016-07-19:
This pins are used in SPI for communicate with flash memory and they are blocked. Nothing to do.
--------------------------------
Usa el buscador antes de preguntar "http://www.esp8266.com/viewtopic.php?f=13&t=3023" y no abras temas cerrados que no se corresponden siquiera. Este sitio de GitHub es para problemas en el código, no para consultar dudas tipo "How to". Para ese tipo de dudas utiliza el foro: www.esp8266.com

Saludos.