esp8266/Arduino #346: ESP8266WebServer and spiffs filesystem

robertgregor, 2015-05-27:
Hello, is there any chance to integrate the ESP8266WebServer and spiffs filesystem together so then it would be possible to e.g. have at the same directory as is sketch the doc directory, where you can put the images / htmls / javascript files and they will be copied to the flash spiffs filesystem and then you could be able to access the files via http? I saw, that it was done for SDcard but there is a lot of space in flash so why not to have it? Or is it possible now? Is there any example how to do it?

igrr, 2015-05-27:
This has already been implemented (a plugin for Arduino to upload files to spiffs was written), but then we removed SPIFFS support, so now this feature is not available.

holgerlembke, 2015-05-27:
For my little project I have written a "filegetter". It accepts simple commands like "format flash", "check flash", "delete file" or can upload files. It is compiled into the arduino-sketch, 140 lines of code.

On the windows side a small C#-console application can feed it. 

I use a cmd-batch to initially upload my stuff into an empty esp, otherwise by command line upload when editing files.

I have no problems putting that into a folder here, but it uses the highly forbidden FS.stuff... :-)

robertgregor, 2015-05-27:
Hello, OK, thanks. My idea was to bundle the web pages together with the sketch, so then during the compilation, it will form just one user bin to upload and that's it. But maybe your idea to upload the files via http is also not bad, can you post somewhere your sketch?

holgerlembke, 2015-05-27:
The bundle concept was my first idea, too. But in practice it is quite annoying. Add/change/fix html/css and you need to recompile, upload, restart etc. and reload in browser.

So I simply up the files and press update in browser. way more convenient.

Have fun: http://lembke.eu/filegetter.zip

robertgregor, 2015-05-27:
Thanks, will try!

rvbcrs, 2015-05-29:
Cool I would love to try it, but I'm fairly new to all this.. If I understand it correctly this is for uploading new files to the flash memory, but how do I serve them when I connect to the module through wifi? Could you please create a simple example?

Thanks!

iamthemadz, 2015-05-29:
I would 100% be wiling to test this, right this instant!

adrianrb, 2015-06-03:
Hello holgerlembke!

I using your FileUpper, and made some improvments, like listing and so.
Files are sent and "seemingly" stored in flash, except...
That when ESP resets they disappear!
I know "that FS API" code will be replaced, but for now it´s my only way to use a file system in ESP.
Could you help me?

All files can be found here: http://adrian-barbosa.net/filegetter.7z

Adrian

holgerlembke, 2015-06-04:
Nice cleanup on the windows side. On the esp8266 side it could be something more modular...

For the problem: I have no clue. You don't do much else than I did. What platform do you use? I use a esp8266-12.

I stopped working on that part of my project to wait what is coming with the new file interface. And as far as my reading into the future goes, I think @igrr is working on something bigger that could change a lot of things/concepts.

adrianrb, 2015-06-04:
Hi Holger, tks for the answer.

My "cleanup" is just a coding preference. I like "C 4 spaces and next line brace", and some newlines separating parts of code, for easy reading, so I use it as standard: in C/C++, Java, Javascript etc. More than 20 years working and many standars after, I´ve stuck with this.

I´ve translated some strings from german to english to make code more "world readable" (I am brazilian).

My ESP is ESP-01. Waitng for a ESP-07 and ESP-12 to arrive.

As I said, I know that @igrr is working on a new code, but I like to keep thinks going on, so I tried your code (didn´t changed much ´cause is simple and works). I like simple things.

How many days to wait for posting?

I´ve looking in NodeMCU sources (and the SPIFFS sources too) to have a clue but didn´t realize why current code doesn´t works.

Ok, I´ll wait for @igrr, or better, continue my other projects...

Oh, if you can, pls suggest me the modularity you quote above: there always space for learning and improvments!

Best wishes!

Adrian

holgerlembke, 2015-06-04:
With modularity I mean: I prefer the solution like in my example: all stuff can live in its own IDE tab. And than there is a setup-call-functions that is called from setup() and (in this case) some sort of worker-function that has to be called from loop().

And I made the getter dynamic, so in case it is not needed, it is not created. (My setup() checks which modules are needed...)

It is some sort of half-way to a real include with all the .h+.c sugar thing. To keep the general overlook I like to split my project in lots of tabs inside the IDE...

I assume @igrr won't care for the error if you report, because it is in FS. and that is banned... :-) If he thinks so, he is totally right. Don't waste time on dead horses.

Perhaps he might outline the future, but I don't care. It is a open source project and he does a lot of work, so THANK YOU is all I can say. Meanwhile I solder my stuff together...

igrr, 2015-08-05:
Implemented in 18b066a, documentation coming up.

tomte76, 2015-09-04:
First of all: Thank you Igrr for all your great work. Can you point me on any documentation how to create and upload a SPIFFS with .js files and using it with your ESP8266WebServer on my modules?

dvukovic, 2015-09-13:
>>removed SPIFFS support

Any idea when SPIFFS will be re-introduced ?

holgerlembke, 2015-09-13:
Current staging has SPIFFS.

Basic documentation is at 

https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/reference.md#file-system

eeijcea, 2015-09-29:
I started working with Lua Framework ... but has a lot of flaws .. this approach looks more stable .. started yesterday and several technical challenges are already solved. 
Thanks Ivan !!
I just dowloaded the Stagging version cause SPIFFS specific need i have ...looking forward to start. Let u know how it goes.

closed 2015-11-06 by Links2004