esp8266/Arduino #347: Error uploading ESP8266 WifiScan Sketch

ranjoy85, 2015-05-27:
Hello everyone,
I am not able to upload WiFiScan Sketch. I am on Arduino 1.6.4
Here is the error

Sketch uses 231,360 bytes (44%) of program storage space. Maximum is 524,288 bytes.
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_sync failed
error: espcomm_open failed

Any help would be highly appreciated.

igrr, 2015-05-27:
and your serial adapter and OS are..?

ranjoy85, 2015-05-27:
I have just updated the firmware ...
now the error is 
warning: espcomm_sync failed
error: espcomm_open failed
OS - Mac 10.10.3
Board - Generic ESP
Programmer = esptool

igrr, 2015-05-27:
You seem to be using a very old version... esptool should not appear in programmers list, it is not necessary to select the programmer now.
What is the USB-serial adapter that you are using?

ranjoy, 2015-05-27:
I was using 1.6.4, but then All tutorials r talking about esptoot. So i somehow Found 1.6.1. M using an arduino to power up. Can u suggest proper steps

ranjoy85, 2015-05-27:
Now I started using 1.6.4 .. 
Board - Generic
Error - Build options changed, rebuilding all 
Sketch uses 198,360 bytes (37%) of program storage space. Maximum is 524,288 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
Attahed sktch
[image]

Hermann-SW, 2015-05-28:
Does any other sketch upload work for you?

I got same error message when having forgotten to connect GPIO0 to GND for flashing. Your photo seems to show 5 cables reaching ESP8266, but 6 are needed -- did you connect GPIO0 to GND for flashing?

Hermann.

Links2004, 2015-05-28:
the reset pin of the Arduino shut be at GND in this case.
otherwise the bootloader from Arduino may corrupt the boot loading to the ESP8266.

ranjoy85, 2015-06-01:
Yes, I figured it out and its working .. :-)

ranjoy85, 2015-06-08:
Getting this message when ever i try to upload my sketch. 
What could be the issue. It seems to be working but all of a sudden it stopped. Is there a way I can reset ESP or upload the boot loader, may sound crazy but anything as such.
 
Sketch uses 198,360 bytes (37%) of program storage space. Maximum is 524,288 bytes.
warning: espcomm_sync failed
error: espcomm_open failed

kevinrhug, 2015-07-17:
Is it possible that the 3.3v pin on the UNO does not prvide enough POWER?  I connected the 5v pin to try to upload and it worked then to run the server I reconnected it to the 3.3v pin.  I think I am going to build a separate source for the ESP8266 that can provide more current at 3.3.. I was wondering if when it was connected to 5v that the arduino could not supply the amount of current necessary and as a result there was a voltage drop.

CptanPanic, 2015-07-19:
Also the tx\rx pins on esp are not 5V tolerant and your arduino is prob using 5V on the serial.

Links2004, 2015-07-19:
you have connected the 5V as VCC to the ESP? 
if yes you have mostly killed your ESP.
the esp require 3.3V and at least 250mA for working, more is better I mostly use a 3.3V 1A LDO or a step down regulator.

kevinrhug, 2015-07-20:
It didn't kill it but it won't maintain the connection to my router for very long. I suppose that the damage may be the reason for this.  Do you program through the Arduino IDE? If so what programmer do you use?

Links2004, 2015-07-20:
do you mean the serial adapter? its an FTDI.
programming works fine with it, including auto reset.
I add an LDO to the 5V of the usb to get stable 3V3 for the ESP.

manojverma75, 2015-08-18:
Here it is the solution for this error. In this video also it was facing the same error but found the solution also
 Have a look to the youtube video
https://www.youtube.com/watch?v=0J1Nt4q7uCQ

knsakib, 2015-09-30:
connect CH_PD to VCC/	3.3V. It should resolve this issue. After connecting turn of and on again. After correcting CH_PD if you see again, power off and on.

narendra93, 2016-01-06:
it is not helping. It's repeatedly showing same error.

SeanBannister, 2016-01-28:
To fix this, I realised I'd selected the wrong board under `Tools... Board...` you'll notice there's a list of ESP8266 boards listed under here.

narendra93, 2016-01-29:
yes, but wont generic module work for basic esp - 07 or esp -01? One more thing, I found out while programming esp-01 that, only blink program uploaded, while others not.

nomanahmed3, 2016-03-05:
had same issue...solution that worked for me....
gpio 0 at gnd permanentrly
when upload start....press reset button

i am using esp12e 
chpd---3.3v
rst----pullup with 10k and push button to pull down momenterily to reset 
gpio 2---3.3v
gpio 15 ---gnd

lrmoreno007, 2016-03-05:
With ESP12, sometime I must hold GPIO0 connected to GND too. I don't know why.

Igor50, 2017-03-06:
WARNING: No widget for SetWidgetProperty command?

BananaAcid, 2018-03-09:
Just for your interest. I solved it, connecting the GND to the USB-Serial-Adapter (not only TX/RX) because I powered the device by my own circuit being powered by a micro-USB power not beeing connected to my PC.
BUT: using the USB power from my PC, as well as connecting the USB-Serial to the same PC, I did not need the GND connected to the USB-Serial (it works with Tx/Rx only).

closed 2018-03-09 by devyte

andreas-bulling, 2019-01-30:
@BananaAcid Thanks a lot for the hint with the GND - that solved my problem and certainly saved hours of further search online!