esp8266/Arduino #348: cbuf.h batch read/write may cause crash

KillingJacky, 2015-05-27:
if(_end < _begin && size_to_read > _bufend - _begin)
in some cases, this statement may be ambiguous, then brings crash.
It should be more explicit:

    if(_end < _begin && (size_to_read > (_bufend - _begin))) 

The same issue lies on write method too.

igrr, 2015-05-27:
Checked the operator precedence table, and ```-``` has higher precedence than ```>```, which has higher precedence than ```&&```. Did you experience a real crash in this method? Can you share the exception info printed to the serial?

KillingJacky, 2015-05-28:
sorry, the reason I pointed is wrong, but the crash happens , when I used the batch write method. Then I wrote a simple test program in loop to test the cbuf:

    cbuf *rx_stream_buffer1;
    rx_stream_buffer1 = new cbuf(256);
    ...
    rx_stream_buffer1->write("1234567890abcdefghijklmnopqrstuvwxyz", 36);  //batch write
    /*const char *str = "1234567890abcdefghijklmnopqrstuvwxyz";
    while (*str++)                         
    {
        rx_stream_buffer1->write(*str);                  //single write
    }*/
   
    while (rx_stream_buffer1->getSize() > 0)
    {
        char c;
        rx_stream_buffer1->read(&c, 1);
        Serial1.println(rx_stream_buffer1->getSize());
        delay(1);
    }

The printed size info will get wrong after a few loops. But if I use the single write and read method, everything works like a charm. I noticed that HardwareSerial implementation only used the single read/write method, so maybe a bug just get ignored.

KillingJacky, 2015-05-28:
Finally I used a ringbuffer implementation here: http://www.asawicki.info/news_1468_circular_buffer_of_raw_binary_data_in_c.html
It works well.

closed 2015-07-30 by igrr