esp8266/Arduino #352: strtok is not working

hariprasadiit, 2015-05-28:
I tried lot of things found in issues to make strtok working,but no luck.

this is my program

#include <string.h>
#include <stdint.h>
#include <stdlib.h>

void setup() {
  
  Serial.begin(115200);
  
  while(1){
     if(Serial.available()){
       char in = Serial.read();
       if(in == 's') break;
       Serial.println("send s");
     }
   delay(10);
   }
  
  
  char rtopic[60] = "hello-LIGHT-L1-POWER";
  char* level;
  char* thing_cat;
  char* thing_id;
  char* feature;

  level = strtok(rtopic,"-");
  thing_cat = strtok(NULL,"-");
  thing_id = strtok(NULL,"-");
  feature = strtok(NULL,"-");

  Serial.println(level);
  Serial.println(thing_cat);
  Serial.println(thing_id);
  Serial.println(feature);
   

}

void loop() {
  // put your main code here, to run repeatedly:

}

its returning the entire string every time.Please point in the right direction if I'm doing something wrong.
same program is working fine with Arduino IDE and Arduino Uno.

Makuna, 2015-05-29:
Investigated:  
strtok calls strtok_r, strtok_r is broken.  
1) It is allocating memory for the returned string when it should not allocate anything. 
2) It should be returning a pointer into the original string and replacing the found token with a null; it is doing neither of these.
3) It is returning the a "reentry state" (third param) that is at the last found token but incrementing past it when it uses to start the next state, thus always returning the same as long as the tokens don't change.

Makuna, 2015-05-29:
https://github.com/esp8266/Arduino/pull/357

igrr, 2015-05-29:
Merged.

closed 2015-06-04 by hariprasadiit

mtnbrit, 2015-07-21:
How can I get this fix please? should it already be in 1.6.5?