esp8266/Arduino #353: continuous watchdog resets with 1.6.4 IDE for ESP8266

Hermann-SW, 2015-05-28:
Hi,

I can compile sketches and upload to ESP8266-01 without problems, but after cutting GND from GPIO0 and resetting my 3.3V UART2TTL connector [1] I see continuous watchdog resets for every sketch I tried. 

The watchdog reset messages can be read in clear at 115200 line speed in IDE Serial Monitor. Therefore I intentionally did setup "my" serial connection at 9600, and added "delay(10)" to most simple sketch:

    void setup() {
      delay(10);
      Serial.begin(9600);
    }

    void loop() {
      delay(10);
      Serial.println(millis());
    }

Most times I do not get a single "millis()" output between 2 watchdog resets, but sometimes some outputs appear before next watchdog reset, see [2]. So the 2 questions I have are:
1. What can be a reason for the continuous watchdogs resets?
2. What can I do in sketch to avoid them?

Hermann.

[1]
http://www.aliexpress.com/item/FREE-SHIPPING-5sets-lot-CP2102-USB-2-0-to-UART-TTL-6PIN-Connector-Module-Serial-Converter/32256355230.html
[2]

    load 0x40100000, len 28780, room����
                                        1�D���H!���x!���B�������@L��)�19b���
                                                                             1�D���)���@H�
          ��DH
    �D���)��
            ��F
    ���?�������?0�ȩ
                   ��F
    �!9D���@L��H!���@H�
                        ��DH
    �D�����F1����D0���
                      ��@L��261
    271
    281
    291
    301
    311
    321
    331
    341
    351
    361
    371
    381
    H!�19�����
              1����H!���
                        ��F
    �!�?���Ό����fL�
                   �
    �!9D���@L��)319�����
                         1@L��266
    276
    286
    296
    306
    316
    326
    336
    346
    356
    366
    376
    386
    396
    407
    417
    427
    437
    447
    457
    467
    477
    487
    497
    507
    517
    527
    537
    H����
         )�F�
    �D�!�����)�19����
                     1�D������T��@H�
                                    !��DH
    ��f�)��)�@H!�1�De�9�D����E)�19���D(���
                                          ��@L���R)���@H�
                                                         !��DH
    ��f��U)���@H�
                 ��DH
    �D�������
             ��F

GerryKeely, 2015-05-28:
Hi
There is no problem with the program.
I would suspect problem is  electrical noise due to an insufficiently rated power supply or noise on the reset pin.Try a different power supply and/or place an electrolytic capacitor(100uF or higher) across the power supply as close as possible to the esp8266 .

Gerry

Hermann-SW, 2015-05-28:
Thank you so much Gerry!

It was noise on reset pin, there are so many conflicting webpages stating which pins need to be connected and how, that I did not connect reset pin at all.

The little program runs for half an hour now without any issues (the small increase by 15 milliseconds instead of 10 in the middle seems to be related to other work done by ESP8266):

    ...
    1801671
    1801682
    1801692
    1801702
    1801712
    1801727
    1801737
    1801747
    ...

Just for completeness, these are the connections between my ESP8266-01 and CH340G USB2TTL capable for  3.3V and 5V.

    ESP8266  USB2TTL (switch on 3.3V)
    VCC      VC
    RESET    VC
    CH_PD    VC
    TX       RXD
    RX       TXD
    GPIO0    [GND for flashing]
    GPIO2    -
    GND      GND

Never thought that wiring could be so easy without breadboard and working. Connect the open yellow cable (GPIO0) to GND (free pin on right connector), press reset on USB2TTL, compile and upload in IDE, open serial monitor and see program running (no disconnect of cable necessary at that point). Only if USB2TTL gets reset or disconnected, disconnect yellow GPIO0 cable from GND, and all is fine for "just running".
 
[image]

Hermann.

closed 2015-05-28 by Hermann-SW