esp8266/Arduino #354: math log() function crashes with reboot.

cornelalexa, 2015-05-28:
Hi,

I have a dew point calculation function which use log() and when used the ESP crashes and reboots.
Here is the code for it.

float dewPointFast(float celsius=0, float humidity=0)
{	
	if(celsius==0 && humidity ==0) return (float) 0.00;
        const float a = 17.271;
	const float b = 237.7;
	float temp = (float)((a * celsius) / (b + celsius));
        temp+= log(humidity/100); //this crashes and reboots
	float Td = (float)((b * temp) / (a - temp));
	return Td;
}

It seems that log10() does the same so it might worth having a look at that too.

Thanks!

cornelalexa, 2015-05-28:
Not sure if it helps, below is the data from crash:

ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x40100000, len 28832, room 16 
tail 0
chksum 0x80
load 0x3ffe8000, len 1564, room 8 
tail 4
chksum 0x8e
load 0x3ffe8620, len 4796, room 4 
tail 8
chksum 0x43
csum 0x43
rd

holgerlembke, 2015-05-28:
Crashes here, too.

Try to live with that:

float log(float dings) {
  static float ln10 = -1;
  
  if (ln10<0) {  
    ln10 = ln(10);
  }  
  return ln(dings)/ln10;
}

float ln(float x) {
  float zw1 = (x-1)/(x+1);
  float zw2 =zw1*zw1;
  float res = 0;
  
  int c=1;
  do {
    res+=zw1/c;
    c+=2;
    zw1*=zw2;
    ESP.wdtFeed();
  } while (zw1>1/1000.0); // resolution...
  
  return res+res;
}

igrr, 2015-05-29:
Same as #233.

cornelalexa, 2015-05-29:
Thank you.

The above functions works as a fix for me. Hope they will be able to fix in the toolchain also. It seems that many functions related to math have issues...

igrr, 2015-06-19:
Fixed in esp8266 branch.

closed 2015-08-05 by igrr