esp8266/Arduino #355: What is a board manager release?

CptanPanic, 2015-05-28:
I read from @igrr in another issue that the board manager will get an update when a new release is made, and I see that the change.md states that the latest release is 1.6.4-673-g8cd3697, whereas the github releases only shows 1.6.1-esp8266-1 as the latest release.  So my questions are; what determines when a board manager update is released? Where does 1.6.4-673-g8cd3697 release name come from? And can we just make all the releases real github releases so it is easier to see when a new release is made.
Thanks.

igrr, 2015-05-28:
I was just writing a reply explaining why I don't use github releases and use arduino.esp8266.com server instead, but then did some googling and found [this](http://www.barrykooij.com/create-github-releases-via-command-line/). Okay, I may go back to github for the next release...

CptanPanic, 2015-05-28:
Great

duncan-a, 2015-05-28:
All lost on me...

Provided the 1.6.4 IDE Board Manager remains correctly synchronised to the
latest 'correctly functioning' version, it can be called anything you like
in my book...!!!

closed 2015-09-30 by igrr