esp8266/Arduino #356: In sdk 1.1    analogWrite(),not work, and 8266 still reset.....

jyzhkj, 2015-05-29:
void setup() {
 
	analogWrite(4,200);
 	Serial.begin(115200);
}

void loop() {
 
	delay(1000);
	Serial.println("1");
}

jyzhkj, 2015-05-29:
wdt reset
load 0x40100000, len 29904, room 16 
tail 0
chksum 0x34
load 0x3ffe8000, len 1364, room 8 
tail 12
chksum 0x8e
ho 0 tail 12 room 4
load 0x3ffe8560, len 1452, room 12 
tail 0
chksum 0x7a
csum 0x7a
s$
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 29904, room 16 
tail 0
chksum 0x34
load 0x3ffe8000, len 1364, room 8 
tail 12
chksum 0x8e
ho 0 tail 12 room 4
load 0x3ffe8560, len 1452, room 12 
tail 0
chksum 0x7a
csum 0x7a
{$

igrr, 2015-07-30:
Fixed in stable package.

closed 2015-07-30 by igrr