esp8266/Arduino #360: word() cast not working

rpseng, 2015-05-30:
Dear all, first let me say that you are doing a great job on this!

The bug I found is very simple to reproduce with the following sketch:

void setup() {
  uint8_t data = 1;
  uint16_t data16 = word(data);
}
void loop() {
}

With the above code I'm getting unresolved symbols in the link step:
undefined reference to `makeWord(unsigned short)'

That should be probably easy to fix, but I don't know where...
I reproduced this problem on both Linux and Windows. For standard Arduino the code links fine.

The following equivalent code links fine with the esp8266:
void setup() {
  uint8_t data = 1;
  uint16_t data16 = uint16_t(data);
}
void loop() {
}

All the best.

chadouming, 2015-05-31:
that's not how you cast in c++ .

this is how:
uint8_t data = 1;
uint16_t data16 = (word)data;

Makuna, 2015-05-31:
See static_cast<>(), dynamic_cast, and reinterpret_cast, const_cast for the modern way.

rpseng, 2015-05-31:
Thanks for the advise:
uint16_t data16 = (word)data;

This indeed also fixes the problem. Anyway, the compiler should complain instead of giving a link problem. This was a real pain to debug, it actually shown up when I was trying to port the ModbusMaster library to esp8266.

All the best.

chadouming, 2015-05-31:
Well, word(something) for the compiler mean you are trying to use the
function word with something argument, so it's normal behavior to say it's
not found.

rpseng, 2015-05-31:
I agree that word(something) means a function call. The problem with that particular case is that the linker complains about 'makeWord' missing and not the original function 'word'. Thus, the compiler somehow understand this, but get confused in the link step... BTW, it works with the standard arduino compiler, the problem appears only with the esp version.

Makuna, 2015-06-01:
I believe the original intent of the makeWord was not specifically for casting, it was meant as a means to abstract "making a word" from two bytes (endian abstraction).

closed 2015-11-06 by Links2004