esp8266/Arduino #362: SPIFFS usage

adrianrb, 2015-05-31:
Hi all.

I´m trying to use the SPIFFS bundled with this project, but stuck.
How use it? Miss a examples folder?

My code:

FSClass fs(0x30000, 0x40000, 256, 64, 5); // how make this values? I´ve put 0x40000 - 0x10000 (64k before API?) sorry i´m a beginner

void setup()
{
    Serial.begin(9600);

    Serial.println("###### Trying create SPIFFS filesystems on address 0x30000 size 64k");
    Serial.println("###### mounting...");
    fs.mount();
    Serial.println("###### formatting...");
    fs.format();
    Serial.print("###### Total bytes: ");
    Serial.println(fs.totalBytes());
    Serial.print("###### Used bytes: ");
    Serial.println(fs.usedBytes()); // <<<<<<<<<< ok till here
    
    Serial.println("###### creating a file...");// >>>>>>>> from here resets over n over...
    char* fname= "test.txt";
    if (!fs.exists(fname)) fs.create(fname);
    
    FSFile f = fs.open(fname, FSFILE_OVERWRITE);
    f.write("testing... line one\r\nline two");
    f.flush();
    f.close();
    
    Serial.print("###### Total bytes: ");
    Serial.println(fs.totalBytes());
    Serial.print("###### Used bytes: ");
    Serial.println(fs.usedBytes());
    fs.umount();
}

Adrian

igrr, 2015-05-31:
My suggestion is — don't use it in it's current form. This API will be
removed in the coming release.

adrianrb, 2015-05-31:
Ohh... But SPIFFS will be bundled with this project in the future or abandoned?
I so, any clue how use flash as FS w/o rewrite an API?
(this removing is related to the issue with SPIFFS creator and/or smings we´ve seen here?)

Adrian

igrr, 2015-05-31:
Yes, rewritten FS is also coming...

adrianrb, 2015-05-31:
thank you. closing...

closed 2015-05-31 by adrianrb

bkrajendra, 2015-07-25:
Any update on new Filesystem library from FalshFS.
Im searching for this solution to store some hardware settings to flash.

seclorum, 2015-08-14:
Is this re-write why FSFILE_OVERWRITE is no longer available in the current FileSystem.h set of mode enums?  If so: why remove the enum?  Very frustrating practice ..