esp8266/Arduino #369: Static IP no DNS possible

cimba007, 2015-06-02:
Might be related to https://github.com/esp8266/Arduino/issues/75

Using static ip I am no longer able to resolve dns-hostnames which is working fine with DHCP. Can somebody confirm this and is there a workaround/solution for this?

Greetings Marc

Links2004, 2015-06-02:
have you set an DNS server in the ESP? 
may ```void dns_setserver(u8_t numdns, ip_addr_t *dnsserver);``` do the jop.

plinioseniore, 2015-06-02:
We got it working just using dhcp and then overriding the ip address with a
static one.

From Mobile.

cimba007, 2015-06-03:
@plinioseniore: I wanted to switch to static IP from the start rather then starting DHCP and then switch it of.

@Links2004: That did the trick

I am currently running an SSD1306 OLED on my ESP8266-ESP3 module. Connecting to my WIFI-Router (Fritz!Box) + aquiring DHCP lease takes quite some time.
```
boot
// doing some stuff like grabbing and parsing web-page, including dhcp-connect
going to sleep after .. 3512ms<\r><\n>
```

I suspected the DHCP lease so I wanted to try out static IP-Addresse but soon found out that DNS was not working.

So I changed following code:
```
void ESP8266WiFiClass::config(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dns)
{
    struct ip_info info;
    info.ip.addr = static_cast<uint32_t>(local_ip);
    info.gw.addr = static_cast<uint32_t>(gateway);
    info.netmask.addr = static_cast<uint32_t>(subnet);

    wifi_station_dhcpc_stop();
    wifi_set_ip_info(STATION_IF, &info);
	
	// Set DNS-Server
	ip_addr_t d;
	d.addr = static_cast<uint32_t>(dns);
	dns_setserver(0,&d);
}
```

and commented out dhcp

```
int ESP8266WiFiClass::begin(const char* ssid, const char *passphrase)
{
...
    //wifi_station_dhcpc_start();
    return status();
}
```

Voila:
```
WiFi.begin(ssid, pass);
WiFi.config(IPAddress(192,168,178,11), IPAddress(192,168,178,1), IPAddress(255,255,255,0),IPAddress(192,168,178,1));
```

```
boot
// doing some stuff like grabbing and parsing web-page, including dhcp-connect
going to sleep after .. 2031ms<\r><\n>
```

Lowering power consumption by a whopping **72%**

PS: If somebody finds this useful and is willing to implement a pull-request this would be very nice. I don't want to download github stuff and begin forking for just this little minor improvement.

chadouming, 2015-06-03:
don't you have to make the config first then begin if you want to use static ip ?

cimba007, 2015-06-03:
As I understand the code it doesn't really matter. The *begin* will onyl connect the your router using the provied ssid/passphrase. In the current code it will also activate dhcp and thus overwrite your static IP. (at least from my experience).

chadouming, 2015-06-03:
nice project, i might need something like this later to work with my wifi thermomether :D

chadouming, 2015-06-03:
Btw, i'm looking into a way to make a clean implementation of your fix

cimba007, 2015-06-03:
Thank you chadouming.

If you need any help later with your project feel free to contact me.

For anyone intrested:

 - OLED Module (SSD1306) (http://de.aliexpress.com/item/Wholesale-0-96-I2C-IIC-SPI-Serial-128X64-White-OLED-LCD-LED-Display-Module-For-Arduino/32235242821.html)
 - Custom PCB (5$ 3-copys, 1square-inch) (https://oshpark.com/)

chadouming, 2015-06-03:
https://github.com/chadouming/Arduino/commit/58f42fe44fc6efeeba7fa864f5861ec033910e2d

This is my current solution. Call config first and dhcp won't be enabled.

chadouming, 2015-06-03:
also, you should take a look at dirtypcbs.com. They have really great pricing and my orders take about 2 weeks from sending the schematic to receiving the board at my home.

drmpf, 2015-06-03:
I tried calling config first the begin() but static IP got lost.  Ended up with DHCP
ESP8266 does not follow Arduino Wifi example in the regard,  I vote for fixing it.

chadouming, 2015-06-03:
my comment was regarding my own patch

cimba007, 2015-06-06:
I would say the issue should be solved:

https://github.com/esp8266/Arduino/commit/b4d483d31ad56448f3c4bb82b43823d96c58d8df

closed 2015-06-06 by cimba007

JanJansen47, 2016-02-24:
I run multiple ESP in a network and want to use fixed IP.  In principle it works however when an ESP is booted for the second time it heads for a different (always the same) IP address.  If I reboot the router it returns the fixed IP address. Any idea? See my code:

```
void connectWiFi(const char* ssid, const char* password, IPAddress ip, IPAddress gateway, IPAddress subnet,   IPAddress dns,  int timeout)
{ int timeoutCounter = 0;
  while (WiFi.status() != WL_CONNECTED) {
    timeoutCounter = 0;
    if (password && strlen(password)) {
      WiFi.mode(WIFI_STA);
      WiFi.begin(ssid, password);
      WiFi.config(ip, gateway, subnet,dns);
      } else {
       WiFi.begin(ssid);
       WiFi.config(ip, gateway, subnet, dns);
    }
    while ((WiFi.status() != WL_CONNECTED) & (timeoutCounter < timeout * 2)) {
      timeoutCounter += 1;
      ::delay(500);
    }
  }
  IPAddress myip = WiFi.localIP();
 if(myip != ip) {ESP.restart();};
}
```

lrmoreno007, 2016-02-24:
I'm using fixed IP with good results. Try in this order and with the delay:

```
  WiFi.config(ip, gateway, subnet, dns);
  delay(100);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
```

Regards

JanJansen47, 2016-02-24:
Hello lrmoreno007,
In my case (ESP8266 and a TL WDR4300 Router from TP Link)  it does not work. I made some variation in the delay but without results. With the order you propose The ESP does NOT get a correct IP address in case I reset the router.   However you TOUCH an issue that might lead to more feedback.
I will try with an other router and see where this goes, Thx anyway

JanJansen47, 2016-02-25:
Hello again,

It's working. I changed the lease time of the router from 120 to 1 minute.  It's working but it's a bit of a miracle....  Maybe it has to do with the special construct  used: first get IP address with the help of dhcp and then change the ip address. Thx for your support

Ir. Jan Jansen
Sagittalaan 22
5632 AL Eindhoven
+31(0)6  888  06529 

  
 

Date: Wed, 24 Feb 2016 04:53:22 -0800

lrmoreno007, 2016-02-25:
In the router try to set a range for DHCP and lets other range for fixed IP. This must be your problem. And you must have a manual count of the fixed IPs you are using, to avoid duplicates.

[image]

In my router:
DHCP start in 10.0.2.30 end in 10.0.2.60
And I use fixed IP starting in 10.0.2.1 and ending in 10.0.2.29. For example 10.0.2.1 is my server, 10.0.2.6 is my PC. For esp IOT i'm really using other subnet 10.0.1.x.

Note: I'm using IPv4 class A (10.xxx.yyy.zzz), and I can make subnets. Surely you are using IPv4 class C (usually in home networks), you only can have a net 192.168.1.xxx, but you can set a range DHCP/fixed IP   equally.

JanJansen47, 2016-02-25:
Hi thanks again,

Yes my fixed IP addresses are also outside the DHCP range. So my fixed IP addresses never coincide with the leased ones.  But that did not solve the problem. My understanding is that this WIFI.begin return already an IP address which is "overwritten" with WIFI.config.  But maybe you have a better idea. See below a copy of procedure.  I'm using "Kolban's book on ESP8266 as my reference. It's quite good and free.  My current interest is to make a robust (survive power and networks outages)  and reliable WiFi service that can replace my distrubuted Java  Enterprise security system.  It's now relying on  a distributed I2C bus across my home.  A very stable and reliable solution but cables are a nightmare.

void connectWiFi(const char* ssid, const char* password, IPAddress ip, IPAddress gateway, IPAddress subnet,   IPAddress dns,  int timeout)
{
  int timeoutCounter = 0;
  while (WiFi.status() != WL_CONNECTED) {
    timeoutCounter = 0;
    if (password && strlen(password)) {
      WiFi.mode(WIFI_STA);
      WiFi.begin(ssid, password);
      WiFi.config(ip, gateway, subnet,dns);      
      } else {
       WiFi.begin(ssid);
       WiFi.config(ip, gateway, subnet, dns);
    }
    while ((WiFi.status() != WL_CONNECTED) & (timeoutCounter < timeout * 2)) {
      timeoutCounter += 1;
      Serial.print(".");
      ::delay(500);
    }
  }
  IPAddress myip = WiFi.localIP();
  Serial.print("IP address: ");
  Serial.println(myip);
 if(myip != ip) {ESP.restart();};
 }

Jan   Jansen  

Date: Thu, 25 Feb 2016 10:51:39 -0800

JanJansen47, 2016-02-26:
Hi,

To get rid of the lease time I disconnect the WiFi in case the requested Static IP does not match the received (DHCP)  IP.  I added the printDiag to see what's happening.  And indeed after the first "wrong" IP the WiFi gets disconnected  and the next time it's a hit.  There might be differences in the routers and it might be an idea to add this type of check in case you want to go static. 

I let it run on all ESP's for some time to see if this turns out to be robust and may post it on a forum. 

void connectWiFi(const char* ssid, const char* password, IPAddress ip, IPAddress gateway, IPAddress subnet,   IPAddress dns,  int timeout)
{ Serial.println("in wifi");
  //Serial.println (ssid); Serial.println (password); Serial.println (ip);
  int timeoutCounter = 0;
  boolean notfound =true;
while (notfound) {
  while (WiFi.status() != WL_CONNECTED) {
    timeoutCounter = 0;
    if (password && strlen(password)) {
      WiFi.mode(WIFI_STA);
      WiFi.begin(ssid, password);
      WiFi.config(ip, gateway, subnet,dns);
         } else {
       WiFi.begin(ssid);
       WiFi.config(ip, gateway, subnet, dns);
    }
    while ((WiFi.status() != WL_CONNECTED) & (timeoutCounter < timeout * 2)) {
      timeoutCounter += 1;
      Serial.print(".");
      ::delay(500);
    }
  }
  delay (1000);
  IPAddress myip = WiFi.localIP();
 Serial.print("IP address: ");
  Serial.println(myip);
 if(myip != ip) {Serial.print("*");WiFi.printDiag(Serial);WiFi.disconnect();}
 if (myip ==ip){notfound=false;}
}
 }

Jan Jansen
Sagittalaan 22
5632 AL Eindhoven
+31(0)6  888  06529 

  
 

From: jan_jansen_47@hotmail.com
To: reply@reply.github.com
Subject: RE: [Arduino] Static IP no DNS possible (#369)
Date: Thu, 25 Feb 2016 23:27:53 +0100

Hi thanks again,

Yes my fixed IP addresses are also outside the DHCP range. So my fixed IP addresses never coincide with the leased ones.  But that did not solve the problem. My understanding is that this WIFI.begin return already an IP address which is "overwritten" with WIFI.config.  But maybe you have a better idea. See below a copy of procedure.  I'm using "Kolban's book on ESP8266 as my reference. It's quite good and free.  My current interest is to make a robust (survive power and networks outages)  and reliable WiFi service that can replace my distrubuted Java  Enterprise security system.  It's now relying on  a distributed I2C bus across my home.  A very stable and reliable solution but cables are a nightmare.

void connectWiFi(const char* ssid, const char* password, IPAddress ip, IPAddress gateway, IPAddress subnet,   IPAddress dns,  int timeout)
{
  int timeoutCounter = 0;
  while (WiFi.status() != WL_CONNECTED) {
    timeoutCounter = 0;
    if (password && strlen(password)) {
      WiFi.mode(WIFI_STA);
      WiFi.begin(ssid, password);
      WiFi.config(ip, gateway, subnet,dns);      
      } else {
       WiFi.begin(ssid);
       WiFi.config(ip, gateway, subnet, dns);
    }
    while ((WiFi.status() != WL_CONNECTED) & (timeoutCounter < timeout * 2)) {
      timeoutCounter += 1;
      Serial.print(".");
      ::delay(500);
    }
  }
  IPAddress myip = WiFi.localIP();
  Serial.print("IP address: ");
  Serial.println(myip);
 if(myip != ip) {ESP.restart();};
 }

Jan   Jansen  

Date: Thu, 25 Feb 2016 10:51:39 -0800

lrmoreno007, 2016-02-26:
It's simple and effective

```
#include <ESP8266WiFi.h>

IPAddress ip(10, 0, 1, 1);  //Will be my fixed IP
IPAddress subnet(255, 0, 0, 0);   //mask
IPAddress gateway(10, 0, 2, 200);  //the router
IPAddress dns(10, 0, 2, 200);  //my router have DNS server

const char* ssid     = "XXXXXXX";
const char* password = "**********";

void connectWIFI(){
  WiFi.config(ip, gateway, subnet, dns);
  delay(100);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED) {
      Serial.print(".");
      delay(200);
  }
  Serial.print("   OK  ---> ");
  Serial.print("IP: ");
  Serial.println(WiFi.localIP());
}

void setup() {
  Serial.begin(115200);
  //Serial1.begin(74880);
  connectWIFI();
}

void loop(){
// do something
}
```

JanJansen47, 2016-02-26:
Thx,

Yes this was my starting point too. However it did not work in my situation.

Jan Jansen
Sagittalaan 22 5632al
+31688806529
It's simple and effective

```
#include <ESP8266WiFi.h>

IPAddress ip(10, 0, 1, 1);  //Will be my fixed IP
IPAddress subnet(255, 0, 0, 0);   //mask
IPAddress gateway(10, 0, 2, 200);  //the router
IPAddress dns(10, 0, 2, 200);  //my router have DNS server

const char* ssid     = "XXXXXXX";
const char* password = "**********";

void connectWIFI(){
  WiFi.config(ip, gateway, subnet, dns);
  delay(100);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED) {
      Serial.print(".");
      delay(200);
  }
  Serial.print("   OK  ---> ");
  Serial.print("IP: ");
  Serial.println(WiFi.localIP());
}

void setup() {
  Serial.begin(115200);
  //Serial1.begin(74880);
  connectWIFI();
}

void loop(){
// do something
}
```

---
https://github.com/esp8266/Arduino/issues/369#issuecomment-189471861