esp8266/Arduino #372: Question: Serial <-> TCP bridge functions

luc-github, 2015-06-03:
Hi,
any recommendation/ source code example about what functions to use to build a bridge
Serial <->TCP

 I plan to use :
```Serial.available() / Serial.read / Serial.write``` for serial part
```ESP8266WebServer::sendContent(String content)``` to send from serial to TCP
but are ```handleClient()``` and  ```void ESP8266WebServer::on(const char* uri, ESP8266WebServer::THandlerFunction handler)``` the best functions to handle data flow from TCP to Serial ? I do not see any function to grab raw data from TCP for a transfert

Any advice ? 
Thank in advance.

drmpf, 2015-06-03:
Check out the example WifiTelentToSerial.
I am putting the finishing touches on a similar sketch that transparently connects Serial to wifi client (bi-directional) 
Also adding pfodWifiConfig (http://www.forward.com.au/pfod/pfodWifiConfig/index.html) support for ESP8266 so you don't have to reprogram to connect to a wifi network or to set the static IP or portNo OR for clients set the host to connect to.

luc-github, 2015-06-03:
Thanks a lot I did not saw this example ... looks perfect

closed 2015-06-03 by luc-github

rogerclarkmelbourne, 2015-10-21:
Hi guys,

I can't find any trace of a WifiTelentToSerial example in the repo

Can you let me know where it is ?

drmpf, 2015-10-21:
You can also look at look at https://github.com/jeelabs/esp-link/tree/v1.0.1.
I am revising http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html
 at the moment.
A couple of problems, 
i) ESP only allows one packet on the fly outbound at a time so this and a 0.2sec ACK limits through put and requires batching to achieve that.  I am adding a 20mS delay on send.  Only send if packet full (say 1300 byte in buffer) OR not incoming Serial data for 20mS.  That seems to work well.
ii) I am currently loosing inbound packets.  The packets are ACKED at the TCP level but being lost between there and the Serial connection.  Still tracking down just where they are going missing.

rogerclarkmelbourne, 2015-10-21:
OK.

Thanks

I just implemented a really simple bidirectional test program, a bit like the Arduinio MegaMultiSerial example, except using WifiClient as one of the serial's. And it seems to generally work OK

I'm using a free terminal program called Hercules which has a TCP server built in.

So I'm pushing data from the ESP rather than using the ESP as the socket server (which would be better)

However at least it allows me to do some remote debug printing, as I am using Serial.swap() as I have another serial device connected to the ESP that I need to communicate with.
But there is a problem when you swap back and forwards as the Serial code changes the GPIO mode and it causes a false char to be sent to the device on the alternative serial pins, which causes it problems.

Hence why I'm trying to do remote Serial prints over TCP