esp8266/Arduino #373: MEM CHECK FAIL!!!

robertgregor, 2015-06-03:
Hello, seems there s a problem with the running of the big sketches. I have sketch which has

Sketch uses 287,824 bytes (66%) of program storage space. Maximum is 434,160 bytes.
Global variables use 49,748 bytes (60%) of dynamic memory, leaving 32,172 bytes for local variables. Maximum is 81,920 bytes.
Uploading 291968 bytes from C:\Users\gregorro\AppData\Local\Temp\build8234347320288677960.tmp/SimpleSwitch.cpp.bin to flash at 0x00000000
..............................................................................................................................................................................................................................................................................................

When I start this sketch I have:

ärlStartrlnnl`nrŽ’’nbŽlr’’nbŽlŒ‚ÕMEM CHECK FAIL!!!
ärlStartrlnnl`nrŽ’’nbŽlr’’nbŽlŒ‚ÕMEM CHECK FAIL!!!
ärlStartrlnnl`nrŽ’’nbŽlr’’nbŽlŒ‚ÕMEM CHECK FAIL!!!
ärlStartrlnnl`nrŽ’’nbŽlr’’nbŽlŒ‚ÕMEM CHECK FAIL!!!
ärlStartrlnnl`nrŽ’’nbŽlr’’nbŽlŒ‚ÕMEM CHECK FAIL!!!

If I load small sketch, everything is fine. What could be the problem?

andyjwang, 2015-06-04:
Set Serial Monitor baud rate at 115200 regardless. These MEM CHECK FAIL!!! messages are normal.

robertgregor, 2015-06-04:
Hello, how they could be normal? The module restarting regulary. I am using 115200 baudrate.

andyjwang, 2015-06-04:
Make sure GPIO 2 is not connected, i.e. float.
Make sure the RX-TX, TX-RX connections are correct. 
In the worst case, flash firmware to your esp8266, then upload your software to esp8266 again.
Yes, it's a struggle. Sometimes you also need to ground RST to either restart it or make it ready for boot loader, or your own software.

andyjwang, 2015-06-04:
Look, I am getting the messages. But it went on.
Connected to COM8 at 115200 baud

Communications Error - Check baud rate

rl
Communications Error - Check baud rate

là|Œlìb|Ž‚ì’r’bŒbŒònnžlnnœâìbpŒŽlrlrl
Communications Error - Check baud rate

pònà‚lbnâ|ìbŒònn
Communications Error - Check baud rate

n
Communications Error - Check baud rate

nl`nrŽ’’nll`pònàrŒœœâàbnâ|ŒlŽrœbŒònn
Communications Error - Check baud rate

lr’’n‚ŽlŒÕMEM CHECK FAIL!!!

Communications Error - Check baud rate

r0In the begining ... Connecting to 

Connecting to OpenWrt3390.2
.........
WiFi connected
Server started
192.168.1.237

sticilface, 2015-06-04:
If ur not powering the ESP from your serial adapter… make sure you connect GND… I had this once… and it looked like that…  you need at least three connections  TX, RX, GND

closed 2015-06-05 by robertgregor

comino, 2015-09-08:
I have the same issue with "MEM CHECK FAIL" and Im using the WROOM. Did you come up with a solution? My sketch is kinda big (but far away from the total flash size) and it doesnt happen with small ones. I didnt have the problem with the E-12. Im powering from my serial sable.