esp8266/Arduino #374: ESP8266FS example or tutorial

robertgregor, 2015-06-03:
Heo, how can I use the java tool ESP8266FS ? Is there any documentation or tutorial or howto document?

me-no-dev, 2015-06-04:
not yet :) requires a binary to deal with creating SPIFFS images, but since support for SPIFFS has been dropped (for now), the tool is useless

robertgregor, 2015-06-04:
I see, but don't understand, why it is dropped, thus this is I think needed very much. I have webserver and want to serve static pages, javascripts, ccs, pictures and so on...

me-no-dev, 2015-06-05:
it's coming, but can not give timeframe. You can hook up an SD card and use that for static content (example available).

robertgregor, 2015-06-08:
I think, there is already the tool for it i.e. spiffy: http://hackersome.com/p/xlfe/spiffy
You download it, create the directory, copy the files and then build it to create the spiff_rom.bin.
I think, here I am not sure, how to burn it to esp, thus if I use 512 flash and 64K for spiffs, which address to use to burn it. And then, there should be support in esp http webserver.

me-no-dev, 2015-06-08:
we have such tool as well and the Arduino tool calls it to do the same thing for the "data" subfolder of the sketch, then uploads it. We had the needed support in Arduino as well and will be back again there soon (hopefully).

igrr, 2015-06-08:
Yeah, https://github.com/igrr/mkspiffs

hallard, 2015-09-25:
Hi guys,
SPIFFS is really, really really a excellent feature, I love it ;-)
in my dreams, a web interface showing FS filesystem in browser with upload/delete/edit files will be the top ;-)

Links2004, 2015-09-25:
you can use this one as base:
https://github.com/esp8266/Arduino/tree/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/examples/SDWebServer

hallard, 2015-09-25:
Damned, you kicked me, I should have read this one before!
But heuuu, sample says I do not have SD, ok I'm kidding ;-)

closed 2015-09-30 by igrr

igrr, 2015-10-01:
@hallard i've updated the fs browser by @me-no-dev for the new FS API, here's the link:
https://files.gitter.im/esp8266/Arduino/jRlu/FSBrowser.zip

Will add this as an example sketch after some cleanup.

hallard, 2015-10-01:
@igrr 
Thanks, works fine, nice Graphs ;-)
I needed to dig into the code to list files on SPIFFS to learn that syntax is something like :
`http://192.168.1.51/edit` ;-)