esp8266/Arduino #381: ESP8266 SMARTCONFIG MODE ON ARDUINO SKETCH

fight182, 2015-06-04:
Hello guys,
Could I get  some  advices?
I got questions about esp8266 during the study.
I want know how to use the smartconfig mode on ESP8266  Arduino sketch.
thank you.

drmpf, 2015-06-04:
I don't use smartconfig (not available on all wifi chips) and has limited functionality and is not open source.

I developed an open source alternative pfodWifiConfig that not only lets you securely connect to your network but also lets you config other settings like DHCP or StaticIP,  server portNo  or for clients the host/port to connect to and the login details, all without reprogramming your module. Your config inputs (via Wifi) are stored in EEPROM

See http://www.forward.com.au/pfod/pfodWifiConfig/index.html  for the details and supporting code libraries for various devices.

Currently putting the finishing touches on the ESP8266 library, should be out later today.

If you don't have an Android (or if your Android mobile does not play nice with Access Points due to your network provider restrictions) you can still do all the config via a computer and telnet.

Since ESP8266 also supports AP mode, I will be adding support to do the config that way in the near future.

fight182, 2015-06-05:
Thank you, drmpf.
I'm learning English. So my English is poor.
I was added pfodWifiConfig Libraries on my Arduino sketch.
But error came during compilation.
Error Message is 
"pfodWifiConfig_ESP8266_PassThrough.ino:32:28: fatal error: pfodWifiConfig.h: No such file or directory
compilation terminated."
I was spent a lot of time in order to solve this error.
But still solution could not be found.
Can you advise about this error to me?

drmpf, 2015-06-05:
Hi sorry you had trouble
" Two supporting Arduino libraries are needed pfodWifiConfig.zip and pfodWifiConfig_ESP8266.zip"
see http://www.forward.com.au/pfod/pfodWifiConfig/ESP8266/pfodWifiConfig_ESP8266.html
for links to these libraries.  Looks like you have not installed pfodWifiConfig.zip

I have just put up a complete tutorial on UART-Wifi Bridge + pfodWifiConfig at
http://www.forward.com.au/pfod/ESP8266/simpleWiFiAddon.html

let me know how you get on.

fight182, 2015-06-05:
Hi drmpf.
Thanks to you i was able to solve error.
But Another error has occurred.
Error message is
"pfodWifiConfig_ESP8266_PassThrough:149: error: 'class pfodWifiConfig_ESP8266' has no member named 'ipStrToNum'".
So I was analyzed "pfodWifiConfig_ESP8266" Header file.
Truly, there was not about "ipStrToNum".
How can i fix this codes?
Thank you.

drmpf, 2015-06-06:
sorry about that, 
web site now updated with latest pfodWifiConfig.zip (including ipStrToNum)

fight182, 2015-06-06:
I was using the ESP-12 module, and followed the pfodWifiConfig's tutorial .
And i was downloaded the pfodWifiConfig ESP8266 Pass through example on my ESP-12.
But it had no response on Arduino's Serial monitor. I guess that it's not working. 
Is this code only works on ESP-01?

drmpf, 2015-06-06:
I have checked the code on ESP8266-01 and on the Olimex ESP8266-EVB board. It work on both so expect it will work on your ESP module.
To get into config mode you need to 
1) apply power then 
2) within 20 sec connect wifiSetup_pin (2) to GND (via a 270ohm resistor for safety)

uncomment 
#define DEBUG  
at the top of the pfodWifiConfig_ESP8266_PassThrough.ino file and you should see 
the 20 sec count down and then 
"Starting pfodWifiConfigV1"
once you gnd wifiSetup_pin

One possible problem is if the pin mapping is different for your module
pfodWifiConfig_ESP8266_PassThrough.ino has 
int wifiSetup_pin = 2; 
// name the input pin for setup mode detection GPIO2 on most ESP8266 boards 

If you replace
if (digitalRead(wifiSetup_pin) == LOW) {
with 
if (1) {

then you will always go into config mode (after 20sec) and you can test the pfodWifiConfig setup.
Once this works you will need to sort out which board pin matches the wifiSetup_pin.  You can change wifiSetup_pin to suit your board.

You can contact me directly at pfod.com.au via support

drmpf, 2015-06-06:
p.s.  On your mobile, once it says Access Point Enabled,  switch to settings on your mobile and check that the access point is actually enabled and has network name of "pfodWifiConfigV1" and your password.  
(I assume you updated the sketch with the new password you generated.)

tzapu, 2015-10-15:
has anyone had more luck with making smart config work ?

tzapu, 2015-10-16:
hi guys, i ve used it like this 
```
  WiFi.mode(WIFI_AP_STA);
  delay(500);

  WiFi.beginSmartConfig();
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    Serial.println(WiFi.smartConfigDone());
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
```
and it worked with android play store app

fight182, 2015-10-17:
I appreciate your reply.

Humancell, 2015-11-02:
tzapu ... a couple of questions about your example code:

1. What version of the code are you using?  Stable?  Staging?
2. What SmartConfig application are you using?  The TI CC3000 application?  You just say "play store app" ... wasn't sure which one.

I've put the code in my sketch and run it, and using the TICC3000 app nothing every happens.  I'm currently running "Stable" code ... but realized tonight that it's actually a long ways back in time, and wondered if I need to use Staging or newer?

tzapu, 2015-11-02:
Hi,
I'm using staging. See here for the app i used http://tzapu.com/2015/10/16/esp8266-smart-config-esp-touch-arduino-ide/

The TI app won't work as that s made for their devices.
Hope it helps

Humancell, 2015-11-03:
Hello,

Thanks for the information on your usage.  I'm going to shift to "staging" and continue to test.

I am unable to use that other app as we need the source to create our own configuration app.  In looking around Github, I found the following links:

https://github.com/EspressifApp/EsptouchForAndroid

https://github.com/EspressifApp/EsptouchForIOS

Is the current version of the libraries for SmartConfig Espressif's Smart Config v2.4?

tzapu, 2015-11-05:
Unfortunately the documentation for all this is a mess. I think the latest SDK has a SmartConfig higher than 2.4. Having said that, Arduino IDE I don't think comes with the latest SDK. 
I used both the android version from the app in play store (no idea what version smartconfig it had) and the ios version built from the github above:
- i was only able to make it work reliably when in line of sight of the router, no walls, 2m away from router, with both phone and esp
- on an Airport Extreme router, even with the above point, most attempts failed, just a few succeeded.

It's not stable enough yet for me at least.

Links2004, 2015-11-06:
the current git version used SDK 1.3.0 (https://github.com/esp8266/Arduino/blob/master/tools/sdk/version)
SDK 1.4.0 has some open problems.

Mirkopr, 2016-12-18:
Hi,
I am having hard time to make working ESP8266 smart config.  I have ESP w/1Mbit flash. The sketch uploaded. Required Port shorted and serial port indicated start of the process but I can not connect nor finished process with web setup http://10.1.1.1.  Please if someone could detail the process...
Thanks

drmpf, 2016-12-18:
hi Mirkopr,
Suggest you ditch 'smart config' and use ESP as an access point with a 'captive portal'
Check out this project for an example http://www.forward.com.au/pfod/HomeAutomation/FanTimer/index.html
does not need any special app on your mobile and works with Apple and Android

Mirkopr, 2016-12-19:
Thanks for suggestion drmpf,
I prefer to have some security as this one looks widely open.
What I am looking is:
uC UART  <-> esp8266<->web

devyte, 2018-02-28:
Closing due to old age.

closed 2018-02-28 by devyte