esp8266/Arduino #384: Serial.println(F("Bla bla bla")); with staging module

luc-github, 2015-06-05:
Hi
Serial.println(F("Bla bla bla")); display Bla bla bla with http://arduino.esp8266.com/package_esp8266com_index.json, so it is working

but it display nothing if I use 
http://arduino.esp8266.com/staging/package_esp8266com_index.json

Is it normal ?  any change in usage of F() macro ?

Thanks,

Makuna, 2015-06-05:
I just checked and this works for me.  I removed all previous packages first.  Then installed a fresh one using the statging, and loaded my test ino, and it displayed the literal string used within a Serial.println.

Do you have your sample ino?

This is what I ran.
```
#include <pgmspace.h>

void setup() {
  // put your setup code here, to run once:
 
  delay(10000); // time to reattach serial debug tool

  Serial.begin(115200);

  Serial.println("Starting print tests");

 
  Serial.println(F("Literal wrapped in F"));

  {
      Serial.print("(String(F())) ");
      String string(F("construction test"));
      Serial.println(string);
      Serial.flush();
  }

    {
        Serial.print("(String =) ");
        String string("doesn't matter");
        string = F("assingment test");
        Serial.println(string);
        Serial.flush();
    }

    {
        Serial.print("(String.concat) ");
        String string("front");
        string.concat(F(" back"));
        Serial.println(string);
        Serial.flush();
    }

    {
        Serial.print("(String +=) ");
        String string("front");
        string += F(" back");
        Serial.println(string);
        Serial.flush();
    }

    {
        Serial.print("(String +) ");
        String stringFront("front");
        String string = stringFront + F(" back");
        Serial.println(string);
        Serial.flush();
    }
    
    {
        Serial.print("(String += blank) ");
        String string("nada");
//        uint8_t i;
//        i = 1;
//       string += (i != 1) ? F(" selected") : ""; // compile error
        string += "";
        Serial.println(string);
        Serial.flush();
    }
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

luc-github, 2015-06-05:
well this is embarassing I cannot reproduce anymore 
my test sketch
```
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
}
void loop() {
Serial.println(F("bla bla bla"));
delay(500);
}
```
when happen I was using only the staging - then moved back to regular to see if issue followed module version and issue disappeared, That is why I raised issue. 
I should come back to staging to confirm issue - my bad, may be this is due to not clean installation from the beginning.

Sorry for the inconvenience and thank for the feedback

closed 2015-06-05 by luc-github

zeeHacker, 2016-03-31:
Can someone help me explain what this code line means: Serial.println(F("Setup code"));
Particularly interested in the F part inside the print function.

philippe-gregoire, 2016-03-31:
The F("string") macro is used to cast the string litteral into a special string stored in Flash memory rather than RAM.
It effectively casts it to const __FlashStringHelper*  using some compiler backdoor directives.
The effect of the cast is that Print::print(const __FlashStringHelper *ifsh) will be called, which pulls the string's chars from flash one by one and writes them to stream.
The regular Print::print(const char str[]) just iterates the in-memory chars and writes them to stream.

So, net result is that this saves on RAM usage, but may be slower.

You can use F() to wrap your own strings, but the methods using it must have a version that accepts a __FlashStringHelper*.

There's a good explanation of this here: http://forum.arduino.cc/index.php?topic=91314.0

Makuna, 2016-03-31:
@philippe-gregoire @zeeHacker  Pretty close, but its really not some compiler backdoor directives; unless you mean the attributes on the type to store the data in code.  
The reason for __FlashStringHelper and the cast to is simple but strange.  The cast is simple to provide a unique type for string data stored in flash, as C++ does not distinguish the pointer type from flash or ram attribute; it considers them the same type.  This unique type is then used by methods so the compiler will link the type to correct methods to call.
Personally I would have wrapped it into a class to get this benefit, but this is how Arduino does it so for compatibility we follow.

philippe-gregoire, 2016-03-31:
Okay, yes, right, I mean here 'backdoor' wrt to the fact that the characterization of the string escapes the standard C++ type expressiveness capabilities.
Among other things, the macro causes the string's storage to be flagged by the compiler as being held in flash and not RAM, directing the linker and archiver to place its bytes in the section that goes in the section of flash that is not mapped/copied to RAM on boot.
Then as you explain, the type into which the string pointer is (forcefully) cast then leverages polymorphism to invoke a method that will know how to handle flash vs directly addressable (RAM) memory...
If that were to be used only from C++, I agree that a class would be nicer (but you would not escape that fact that this storage has to be placed in the specific section of flash that stays in flash/SRAM and is not copied to the mcu's volatile operating RAM on boot.).

Makuna, 2016-04-01:
And adding some complexity, there is one ARM compiler (used for Arduino) that when it sees const on a variable, it by default treats it as PROGMEM deviating from starndards.
Another wish might have been that C++ would treat the attributes on the type as unique types.

zeeHacker, 2016-04-02:
Thank you very much Philippe Gregoire and Michael Miller for the
elaborate details.