esp8266/Arduino #385: TX problems over wifi

kdschlosser, 2015-06-05:
extremely slow, tried several different ESP's and different IDE's the TX is like 30 CPS. any suggestions would be helpful. RX is fine 5000 +cps (telnet application) realterm and esp, tried 3 different computers same results,

me-no-dev, 2015-06-05:
the way you send data is probably wrong. Mind posting that part of your sketch so I can help you? You are sending a packet with each char and that is your problem.

kdschlosser, 2015-06-05:
ok as a test i did this. I understand what you mean by sending a packet at a time. i am not using the write command and again this is as a test. but i pretty much use the println with a series of bytes.
not a big fan. sketch i wanted to run on it is a couple thousand lines of code, not going to post that whole thing on here.
but this does the exact same result even if i push a test file through it still slow only on transmit

#include <ESP8266WiFi.h>

const char* ssid     = "blah blah";
const char* password = "";

const char* host = "123.456.789.000";
int port = 8125;

WiFiClient wifiClient;

void setup() {
	Serial.begin(384000);
	delay(10);
	Serial.println();
	Serial.println();
	Serial.print("Connecting to ");
	Serial.println(ssid);
    WiFi.begin(ssid, password);
	while (WiFi.status() != WL_CONNECTED) {   //<------- and this causes my ESP to
		delay(500);                           //causes my ESP to boot loop if it doesn't get an IP
		Serial.print(".");                     //DHCP has nothing to do with wifi connectivity
	}
	Serial.println("");
	Serial.println("WiFi connected");  
	Serial.println("IP address: ");
	Serial.println(WiFi.localIP());

	Serial.print("connecting to ");
	Serial.println(host);
	if (!wifiClient.connect(host, port)) {
		Serial.println("connection failed");
		return;
	}
}

void loop() {
	while (wifiClient.available() > 0) {
		byte inByte = wifiClient.read();
		Serial.write(inByte);
	}
	while (Serial.available() > 0) {
		byte inByte = Serial.read();
		wifiClient.write(inByte);
	}
}

me-no-dev, 2015-06-05:
here is your problem :)
```cpp
while (Serial.available() > 0) {
byte inByte = Serial.read();
wifiClient.write(inByte);
}
```

change this to:
```cpp
size_t len = Serial.available();
char buf[len];
for(size_t i = 0; i < len; i++) buf[i] = Serial.read();
wifiClient.write(buf, len);
```

igrr, 2015-06-05:
```C++
size_t len = Serial.available();
char buf[len];
```
this is not valid code unfortunately :)

me-no-dev, 2015-06-05:
why not?

igrr, 2015-06-05:
Okay, in this seems to be a GCC extension:
https://gcc.gnu.org/onlinedocs/gcc-4.1.2/gcc/Variable-Length.html#Variable-Length

You actually need to turn on ```-pedantic``` to make GCC issue an error
```
buffer.cpp:9:17: warning: ISO C++ forbids variable length array 'buf'
[-Wvla]
char buf[rand()];
```

Looks like I learned something today.

igrr, 2015-06-05:
Reading about the implementation, these arrays are allocated on the stack. This is not a very good practice with on ESP because we have a short 2Kb stack for the arduino thread... So depending on the amount of data received having a static array or allocating it on the heap may be better.

me-no-dev, 2015-06-05:
what's the correct way to create a buffer in that case then?
using a buffer is a must, else he transmits just a char in each packet.

Links2004, 2015-06-05:
some think like this:
 ```
        size_t len = Serial.available();
        if(len) {
            char * buf = (char *) malloc(len);
            if(buf) {
                Serial.readBytes(buf, len);
                wifiClient.write(buf, len);
                free(buf);
            } else {
                //too less memory left
            }
        }
```

me-no-dev, 2015-06-05:
thanks Markus :)

igrr, 2015-06-05:
Better yet (dynamic allocation is slow):

```C++
static const bufferSize = 128;
static const buffer[bufferSize];

size_t len = stream.available();
while (len) {
    size_t will_copy = (len < bufferSize) ? len : bufferSize;
    stream.readBytes(buffer, will_copy);
    client.write(buffer, will_copy);
    len -= will_copy;
}
```

drmpf, 2015-06-06:
This version of the code actually compiles

static const size_t bufferSize = 128;
static uint8_t sbuf[bufferSize];

  //check UART for data
  if (Serial.available()) {
    size_t len = Serial.available();
    while (len > 0) { // size_t is an unsigned type so >0 is actually redundent
      size_t will_copy = (len < bufferSize) ? len : bufferSize;
      Serial.readBytes(sbuf, will_copy);
      //push UART data to connected client
      if (alreadyConnected) {
        client.write((const uint8_t *)sbuf, will_copy);
        delay(0); // yield
      }
      len -= will_copy;
    }
  }

kdschlosser, 2015-06-11:
Ok. What exactly out of all of this actually works?. And a note about this should be included in any documentation or even better yet an example sketch. I do understand now where I was not doing correctly. But because of stack limitations and such the absolute best was to a achieve proper TX speeds with the esp module should be included in some kind of an example. There are a lot of people out there that are new like I am as you could probably tell. I would have never even looked at the fact I was only sending one byte at a time in one packet. And another question. Is the statement " this compiles"  I take that as it compiles but have not actually written it to an esp to see if it works. Don't know if I am correct in that or not. But working to me is it compiles and have tested it

kdschlosser, 2015-06-11:
I am thinking that maybe some kind of begin and end like i2c would be easier and having it built into the library if there is a known limitation. I know that sounds hokey but that is the purpose to libraries. To simplify coding. I thought as I am sure many did that this functioned the same way as the ethernet or WiFi libraries that are already built into the arduino ide. Because I have not an issue with either one of those with the way the code was written beforehand.

me-no-dev, 2015-06-11:
AVR and ESP8266 are really really different devices in many aspects. This is a new platform to all and we are learning as we go. We have implemented helper methods where possible (web server for instance).
For the stack to know how much data you are going to send, so it can chunk and packetize it properly, that data needs to be given. Calling connection.write(byte) can in no way give that information, therefore it sends it right away.

drmpf, 2015-06-11:
I had the same problem with .write(byte).  I implemented a Print class the Arduino way by just over overwriting write(byte)  but it was sooooo slow.
Went back to direct write(buffer... ) to wifi client. 
much better.

kdschlosser, 2015-06-11:
First off. I really think the work that is getting done is awesome. And in by no means am I trying to put any of that down. I do thank all of the people that have contributed to this project.

me-no-dev, 2015-06-12:
no worries :) we are all trying to get this thing working:)

av1024, 2015-07-02:
Have similar issue on some telnet clients. E.g. All ok on ubuntu/mint-x64, but not on debian/lxde-armv7 or android terminal.

diagnostic code from WiFiClient::wrtite() (thanks @igrr ) show that some strings printed char-by-char. Removing F() macro resolves char-by-char issue
```C
prn.println(F("Try 'help cmd' for detailed help\n\r"));
   prn.println(F("Allowed commands:"));
   
    for(int i=0;i<CLI_MAX_COMMANDS;i++) {
      // _cmds[i] == String (F("some text") added via cli.add(String, String, String, void*):
      // from cli.add("print", F("Print all sensor values"), F(""), on_singleWord);
      if (_cmds[i]._name != "") {
        prn.print(_cmds[i]._name);
        prn.print("\t\t");
        prn.println(_cmds[i]._short);
      }
    }
```
Diagnostic prints from WiFiClient::write(buf, size):
```
wifi write: 1 (T)
wifi write: 1 (r)
wifi write: 1 (y)
wifi write: 1 ( )
wifi write: 1 (')
wifi write: 1 (h)
wifi write: 1 (e)
wifi write: 1 (l)
wifi write: 1 (p)
wifi write: 1 ( )
wifi write: 1 (c)
wifi write: 1 (m)
wifi write: 1 (d)
wifi write: 1 (')
wifi write: 1 ( )
wifi write: 1 (f)
wifi write: 1 (o)
wifi write: 1 (r)
wifi write: 1 ( )
wifi write: 1 (d)
wifi write: 1 (e)
wifi write: 1 (t)
wifi write: 1 (a)
wifi write: 1 (i)
wifi write: 1 (l)
wifi write: 1 (e)
wifi write: 1 (d)
wifi write: 1 ( )
wifi write: 1 (h)
wifi write: 1 (e)
wifi write: 1 (l)
wifi write: 1 (p)
wifi write: 1 ( 0A)
wifi write: 1 ( 0D)
wifi write: 2 ( 0D 0A)
wifi write: 1 (A)
wifi write: 1 (l)
wifi write: 1 (l)
wifi write: 1 (o)
wifi write: 1 (w)
wifi write: 1 (e)
wifi write: 1 (d)
wifi write: 1 ( )
wifi write: 1 (c)
wifi write: 1 (o)
wifi write: 1 (m)
wifi write: 1 (m)
wifi write: 1 (a)
wifi write: 1 (n)
wifi write: 1 (d)
wifi write: 1 (s)
wifi write: 1 (:)
wifi write: 2 ( 0D 0A)
wifi write: 5 (print)
wifi write: 2 ( 09 09)
wifi write: 23 (Print all sensor values)
wifi write: 2 ( 0D 0A)
```

closed 2015-11-06 by Links2004

heroseller17, 2016-05-27:
Hello i have a problem with ESP8266 thats ...sometimes it works great but randomly it seems dont responding at TCP conection but my code is still working! because i always send a char on serial port and i have that char when TCP conection is lost!why?

marioelectron, 2021-05-03:
in WiFi.h     setTimeout  in ESP32 not setup 

uint32_t timeout = (1);
      user = i;

///uint32_t latency =1 * 10;
serverClients[i].setTimeout(timeout);  ///not work?
  
serverClients[i].write((uint8_t *) &PString, sizeof(PString.length()));

drmpf, 2021-05-03:
Danixsadati 
Could be a half Open connection, see
[Detection of Half-Open (Dropped) TCP/IP Socket Connections](http://www.codeproject.com/Articles/37490/Detection-of-Half-Open-Dropped-TCP-IP-Socket-Conne)
I put a timeout in my code and I stop getting new data I close the connection on my side after the timeout