esp8266/Arduino #386: ESP8266WifiMulti bugs

av1024, 2015-06-05:
As far as I know C, there is several bugs in ESP8266WiFiMulti::APlistAdd:

```c
bool ESP8266WiFiMulti::APlistAdd(const char* ssid, const char *passphrase) {

    WifiAPlist_t newAP;

    //!!! check for NULL/empty ssid required (causes "malloc(0) error" on empty strings)
    //+++
    if (!ssid || !*ssid) return false
   
    newAP.ssid = (char*) malloc(strlen(ssid));  // !!! should be strlen+1

    if(!newAP.ssid) {
        return false;
    }

    strcpy(newAP.ssid, ssid);

    if(passphrase && *passphrase != 0x00) {
        newAP.passphrase = (char*) malloc(strlen(passphrase));  // !!! should be strlen+1
    }

    // looks like never add SSID with empty password
    if(!newAP.passphrase) {
        free(newAP.ssid);
        return false;
    }

    strcpy(newAP.passphrase, passphrase);

    APlist.push_back(newAP);
    return true;
}
```
BTW, I think it is not good idea to rescan wifi points every time. `run()` freezes for several seconds if no valid SSID available

Links2004, 2015-06-05:
fixed, if the freezes is good or bad is depends on the application.
have some ideas to run the scan in background but for this some changes on the ```ESP8266WiFiClass``` class are needed. for this some free time to test this out a needed.

av1024, 2015-06-05:
Hmm...
`154: if(!ssid || strlen(ssid) > 31`: You still allow to pass empty SSID here

using `strdup(ssid)` instedad of strlen+malloc saves 16bytes per call for me and looks potentially a bit faster (http://www.vanheusden.com/misc/strdup/strdup_v_SlMalMemc.php)

About freezes: (as an idea) For now I wrote "alternate" version of wifiMulti - only cycle between added SSID's without scanning and RSSI analyzing. The simple use case is: two known AP's, for example at home and at work. So I do need only connect to known SSID, not to select th best ))

Links2004, 2015-06-05:
the compare of time is for an x68 cpu. the implementation on the ESP is like (strlen + malloc + memcpy):
https://github.com/esp8266/Arduino/blob/7dc997b4a451e7ef0823be07ca379c31f805639a/hardware/esp8266com/esp8266/cores/esp8266/libc_replacements.c#L245

so faster i not think so, but it will save some bytes yes :)

yes you can try "blue forcing" the right AP. but you use case is much simpler the one i have.
I have 5 different WiFi on 3 different locations with multiple APs. so scanning ones if faster the try 5 times a may possible WiFi. The ESP do a scan internally be WiFi connect.

*coding half sleeping is a bad idea*

commit d0315a33 (2015-06-05): fix #386
commit 078efcb9 (2015-06-05): fix #386
closed 2015-06-05 by igrr