esp8266/Arduino #387: Support for NodeMCU v1.0 dev kit (with ESP-12E module)

knovoselic, 2015-06-05:
There's a newer version of NodeMCU devkit board:
https://github.com/nodemcu/nodemcu-devkit-v1.0

It has ESP-12-E module and 32M of flash storage. Would it be too much work to make it work with Arduino IDE?

I've been messing with boards.txt, but then I've figured out that I need eagle.flash.32m.ld. Where can I find that file?

igrr, 2015-06-05:
Nodemcu 1.0 is supported: https://github.com/esp8266/Arduino/blob/esp8266-sdk-1.0/hardware/esp8266com/esp8266/boards.txt#L235
In fact this is the board i'm using right now. Regarding 32M — is it really 32MB or 32Mb (4MB)?

marvinroger, 2015-06-05:
[It is an open hardware, with ESP-12-E core [32Mbits(4MBytes) flash version].](https://github.com/nodemcu/nodemcu-devkit-v1.0#nodemcu-devkit-v10)

Makuna, 2015-06-05:
Sorry I can't link the other issue, there is a change staged to the boards.text to support V2 which is basically a 12e, so it may not be working until the the next release or you make a small change in this text file.

knovoselic, 2015-06-06:
Thanks guys, I got little confused with 32M. The board is 32Mbit, 4MB. I've copied NodeMCU v1.0 config and now it's working. Thanks again!

closed 2015-06-06 by knovoselic

lcltech, 2015-06-16:
If I used the boards.txt link to by @igrr  flashstringhelper do not work.
 http://arduino.esp8266.com/package_esp8266com_index.json
Does not include nodemcu V1.0, any idears?

Makuna, 2015-06-16:
Things are moving fast, the released boards install is getting stale.  There is a staging one but use at your own risk.

http://arduino.esp8266.com/staging/package_esp8266com_index.json