esp8266/Arduino #391: AnalogRead reboot micro

Testato, 2015-06-06:
After a 3 or 4 analog reading the chip reboot and on serial i receive this message.

Code:
valAnalog=analogRead(A0);
Serial.println(valAnalog); 

Error:
----------------------------
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 28780, room 16 
tail 12
chksum 0x2d
ho 0 tail 12 room 4
load 0x3ffe8000, len 1332, room 12 
tail 8
chksum 0xc6
load 0x3ffe8540, len 1576, room 0 
tail 8
chksum 0xa4
csum 0xa4
------------------------------------

wizard23, 2015-06-17:
do you have a while loop around the measurement and writing of the analog value? If so make sure to to call delay(0) in the loop to give the os backround tasks a chance to execute otherwise the watchdog timer will cause a reset (wdt reset) because it thinks your programm is not functioning correctly. One of this copoerative background tasks signals the watchdog that it does not have to reset the device.

If thats not it include the full source otherwise it's hard to diagnose the problem ;)

chadouming, 2015-06-17:
Add delay(10); after the analog read.

Testato, 2015-06-18:
i have no while or delay inside, only a simple read, i think this Bug is a duplicate of this issue.
I writed here what i discovered on offical Api documentation
https://github.com/esp8266/Arduino/issues/338#issuecomment-113099775

chadouming, 2015-06-18:
> #338 (comment)

Try to add the delay right after the reading. This fixed my problem.

mvdbro, 2015-06-20:
Same issue here. When in AP mode, no problems. When connected in STA mode and reading ADC, ESP resets after one try... Funny...

Temp fix to revert to an older function call by changing "wiring_analog.c"

from this:
  return readvdd33() >> 2; // readvdd33 is 12 bit
to this:
  return system_adc_read();

Testato, 2015-06-20:
Is there a pull request on this point. Try it and wrute your feedback.
Thanks

>
> Same issue here. When in AP mode, no problems. When connected in STA mode
and reading ADC, ESP resets after one try... Funny...

closed 2015-07-30 by igrr