esp8266/Arduino #394: Including wifi headers in a headerfile.

Duality4Y, 2015-06-06:
Hi there, 
How do you include forexample:
#include <ESP8266WiFi.h>
#include "WiFiUdp.h"

into a external headerfile ?
because if i do that it tells me this:

Arduino: 1.6.5 (Linux), Board: "Generic ESP8266 Module, 160 MHz, 460800, 4M (3M SPIFFS)"

In file included from /home/robert/arduino-1.6.5/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:38:0,
                 from Artnet.h:28,
                 from Artnet.cpp:25:
/home/robert/arduino-1.6.5/hardware/esp8266com/esp8266/cores/esp8266/esp8266_peri.h:379:0: warning: "SPIBUSY" redefined [enabled by default]
 #define SPIBUSY  (1 << 18) //SPI_USR
 ^
/home/robert/arduino-1.6.5/hardware/esp8266com/esp8266/cores/esp8266/esp8266_peri.h:281:0: note: this is the location of the previous definition
 #define SPIBUSY   9 //wait SPI idle
 ^
In file included from Artnet.cpp:25:0:
Artnet.h:30:25: fatal error: ESP8266WiFi.h: No such file or directory
 #include <ESP8266WiFi.h>
                         ^
compilation terminated.
Error compiling.

wizard23, 2015-06-09:
My guess is that you don't include the required header files in the main sketch. In the build process only the header files referenced in the main sketch will be available. This is weird for people that are used to normal C/C++.

so try including ESP8266WiFi.h and WiFiUdp.h at the beginning of zour main sketch.

to quote from the official arduino doc: "The include path includes the sketch's directory, the target directory (<ARDUINO>/hardware/core/<CORE>/) and the avr include directory (<ARDUINO>/hardware/tools/avr/avr/include/), as well as any library directories (in <ARDUINO>/hardware/libraries/) which contain a header file which is included by the main sketch file."
http://www.arduino.cc/en/Hacking/BuildProcess

Duality4Y, 2015-06-10:
maybe i understand or maybe I don't,
but I find this not useful at all

duncan-a, 2015-06-10:
I find it ideal to have a header file for a common setup...

Duality4Y, 2015-06-10:
i see your point,
I wanted to have my code all nice together in a separate source file and it's dependencies included in it.

closed 2015-07-30 by igrr