esp8266/Arduino #398: Min max, not working

sticilface, 2015-06-07:
I was using min and max just fine in a previous build.  but the latest pull it has stopped working.

````

b_hsv.ino: In function 'double threeway_max(double, double, double)':
b_hsv:176: error: 'max' was not declared in this scope
b_hsv.ino:176:27: note: suggested alternative:
In file included from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/memory:62:0,
                 from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h:28,
                 from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:32,
                 from Melvana.ino:4:
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h:260:5: note:   'std::max'
     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^
b_hsv:176: error: 'max' was not declared in this scope
b_hsv.ino:176:28: note: suggested alternative:
In file included from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/memory:62:0,
                 from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h:28,
                 from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:32,
                 from Melvana.ino:4:
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h:260:5: note:   'std::max'
     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^
b_hsv:176: error: redeclaration of '<typeprefixerror>max'
b_hsv.ino:176:27: note: previous declaration '<typeprefixerror>max'
b_hsv.ino: In function 'double threeway_min(double, double, double)':
b_hsv:180: error: 'min' was not declared in this scope
b_hsv.ino:180:27: note: suggested alternative:
In file included from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/memory:62:0,
                 from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h:28,
                 from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:32,
                 from Melvana.ino:4:
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h:239:5: note:   'std::min'
     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^
b_hsv:180: error: 'min' was not declared in this scope
b_hsv.ino:180:28: note: suggested alternative:
In file included from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/memory:62:0,
                 from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h:28,
                 from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:32,
                 from Melvana.ino:4:
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h:239:5: note:   'std::min'
     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^
b_hsv:180: error: redeclaration of '<typeprefixerror>min'
b_hsv.ino:180:27: note: previous declaration '<typeprefixerror>min'
Exception in thread "Thread-17" java.lang.Error: Error: could not match input
	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.zzScanError(CPlusPlusTokenMaker.java:2077)
	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.yylex(CPlusPlusTokenMaker.java:2311)
	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.getTokenList(CPlusPlusTokenMaker.java:1901)
	at org.fife.ui.rsyntaxtextarea.RSyntaxDocument.getTokenListForLine(RSyntaxDocument.java:416)
	at org.fife.ui.rsyntaxtextarea.SyntaxView.getLineWidth(SyntaxView.java:302)
	at org.fife.ui.rsyntaxtextarea.SyntaxView.calculateLongestLine(SyntaxView.java:110)
	at org.fife.ui.rsyntaxtextarea.SyntaxView.updateMetrics(SyntaxView.java:892)
	at org.fife.ui.rsyntaxtextarea.SyntaxView.setSize(SyntaxView.java:802)
	at javax.swing.plaf.basic.BasicTextUI$RootView.setSize(BasicTextUI.java:1722)
	at javax.swing.plaf.basic.BasicTextUI.getPreferredSize(BasicTextUI.java:912)
	at javax.swing.plaf.basic.BasicTextAreaUI.getPreferredSize(BasicTextAreaUI.java:120)
	at javax.swing.JComponent.getPreferredSize(JComponent.java:1662)
	at javax.swing.JTextArea.getPreferredSize(JTextArea.java:619)
	at javax.swing.JViewport.getViewSize(JViewport.java:999)
	at javax.swing.plaf.basic.BasicScrollPaneUI.syncScrollPaneWithViewport(BasicScrollPaneUI.java:278)
	at javax.swing.plaf.basic.BasicScrollPaneUI$Handler.stateChanged(BasicScrollPaneUI.java:1033)
	at javax.swing.JViewport.fireStateChanged(JViewport.java:1369)
	at javax.swing.JViewport.setView(JViewport.java:969)
	at javax.swing.JScrollPane.setViewportView(JScrollPane.java:1007)
	at org.fife.ui.rtextarea.RTextScrollPane.setViewportView(RTextScrollPane.java:253)
	at processing.app.Editor.setCode(Editor.java:1818)
	at processing.app.Sketch.setCurrentCode(Sketch.java:998)
	at processing.app.Editor.statusError(Editor.java:2784)
	at processing.app.Editor$BuildHandler.run(Editor.java:2062)
	at java.lang.Thread.run(Thread.java:745)
```

Makuna, 2015-06-09:
if you include ESP8266WiFiMulti.h, this will undef min and max macros.
This is done due to the stl classes it uses having method names of min and max and the preprocessor would replace them with the macro and fail to compile.

Lenbok, 2015-08-07:
Is this fixed yet?

I had this problem with 1.6.5, and also a related one when I try to use the AdaFruit GFX library, there is a conflict with the definition of swap(). It looks like the new ESP libraries are clobbering stuff that just worked before.

seclorum, 2016-05-03:
Hi .. I recently ran into this problem - is there any reason why the macro's can't be undef'ed in Arduino.h by default, and only defined if STL is not present?  Seems like the guard for this check should be in Arduino.h, so that we don't have to pepper #undef's all over the place further downstream ..

igrr, 2016-05-03:
That's an awesome idea, but how would you implement this? I.e. how would you check in Arduino.h whether there is an STL header included somewhere later?

seclorum, 2016-05-03:
Ah, good point. :)

Makuna, 2017-03-26:
The key issue here is that macros that use common names are a bad technique to use.  Arduino standards should replace these with inline functions and this would solve the problem.

devyte, 2017-10-20:
@igrr @Makuna does anyone have an idea for how to fix this? Is this still an issue in the standard Arduino? If not, how did they fix it?
Would migrating to a newer gcc help?

Makuna, 2017-10-20:
With AVR, it is also a macro, but it is less of an issue as they don't support STL at all.  If you do use STL (there are a few libraries for it but are missing major features) you run into the same problem.

devyte, 2017-11-15:
@sticilface @Lenbok @seclorum there was a PR merged about this, Could you please retest with latest git?

s0170071, 2017-12-20:
ran into the max() problem compiling git version of ESPEasy with the git version of the core.

Diaoul, 2017-12-28:
In the meantime what is the proper way to have Arduino and ESP8266 compatible code?

s0170071, 2017-12-28:
Check this out:

https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=3896&p=21747&hilit=Max#p21747

rgmorrison, 2018-01-05:
^ also having this exact issue with Adafruit MQTT library. Thank you for the fix.

tbam, 2018-01-05:
I was also having this issue with Adafruit MQTT library. Thank's for the fix!

devyte, 2018-01-08:
To everyone encountering build issues in 3rd party libs due to min|max:
The implementation of the Arduino core in this repo has migrated to conform to the C++ standard of min|max usage. The C++ standard min/max are not #defines, but templated functions that force the user to pass in both arguments of the same type. This is on purpose, and differs from the old-style #defined min|max implementation that is common in Arduino, where you can pass any pod type in.
The C++ standard decided on same-type args, because the #defined way made for implicit bugs, user errors, and build warnings with the usage.
For those 3rd party libs that have issues building, the reason is precisely that at some points in their code there are arguments of different types being passed in, e.g.: min(a,b) where a is int and b is uint16.
The 3rd party authors need to fix all such cases. In most, it is enough to cast one of the arguments being passed in to the type of the other, e.g.: min(a,(uint16)b). They will have to think about which way to case, and that is a good thing.
In other cases, they'll realize there is actually a bug in their usage, or even redundant code, and will need to fix it. That is also a good thing.
The above means that this is not a bug in this repo, but a fix in this repo that is exposing potential bugs in those libs, because our code is now stricter. Thanks to the C++ way, those exposed bugs are now caught at compile time instead of at run time, which is always desirable.
I suggest opening issues in the repos of those 3rd party libs requesting fixes. In the meantime, you can apply a workaround like the one described above if you really need it, but I advise against it.

I consider this resolved, so closing.

closed 2018-01-08 by devyte

ArfyFR, 2018-01-08:
Hi, for information Adafruit_MQTT_Library library did a fix:
https://github.com/adafruit/Adafruit_MQTT_Library/issues/106
merge:
https://github.com/adafruit/Adafruit_MQTT_Library/pull/107

But I didn't see a release yet.

devyte, 2018-01-08:
The referenced fix in their repo is not the right way sigh

ArfyFR, 2018-01-09:
Sorry, I only replied with the URL inside the text.  It was marked as reference by github I suppose.

ladyada, 2018-01-10:
hi if there's a better fix please submit a PR and we'll merge it in!

devyte, 2018-01-10:
@labdaya I already explained what needs to be done above when I closed this. I am not a user of the referenced adafruit repo, or even of mqtt, so asking me for a PR is unrealistic. My interest is in improving this repo as a collaborator, and making correct use of C++ is basic in that.
The build errors should provide you good hints of what's wrong on a case by case basis.

ladyada, 2018-01-10:
@devyte sorry, i'm still very confused, the original code had a cast so both were uint16_t, but that wasnt working, so we replaced with a ternary which does the same thing. so what is wrong with the PR we merged alreaedy? Thanks for your help!!!

s0170071, 2018-01-10:
@develo: your solution worked for me. Thank you for explaining.

https://github.com/esp8266/Arduino/issues/398#issuecomment-356498540
Diese Nachricht wurde von meinem Android-Gerät mit K-9 Mail gesendet.

jerryneedell, 2018-01-10:
@devyte I am still having a hard time understanding what you are suggesting for the correct what to use min().  Can you tell me what is wrong with this statement and how you would rewrite it?
 len is a uint16_t
uint16_t sendlen = min(len, (uint16_t)250);

Thank you.

tofrnr, 2018-02-25:
I don't understand.

My  code is (board version 2.3.0, nodeMCU 1.0, IDE 1.8.5):

`unsigned long test = min( millis()-100*1000ul, millis() );`

and the error is 

> min is not declared in this scope

. Nonetheless, "min" is orange highlighted in the editor.

So please, in simple words: what the heck is wrong?