esp8266/Arduino #399: Starting additional Tasks

0xPIT, 2015-06-07:
(How) is it possible to start FreeRTOS tasks?
When I try to start a task from setup() what increments an integer every 5s,
the Target resets after a couple of seconds of shows rather strange behavior.

Any suggestion / sample, how to create a task?

igrr, 2015-06-08:
This project is not based on FreeRTOS, so you have to use Espressif's SDK
calls to start a task. Make sure you don't call any blocking functions from
this new task as it is cooperatively scheduled. The task function has to
return to avoid blocking execution of other tasks. Calling certain Arduino
functions (delay, Serial IO, WiFiClient IO) may also cause unexpected
behavior.

0xPIT, 2015-06-08:
Thanks for your repliy. 

Oh sorry, I thought the SDK is based FreeRTOS.

So, I receive the behaviour I've reported above by copying the code for the `loop_task`.

I've created a global `volatile uint32_t`, and in my `testTask`, I loop and assign the current ticks to it, then pause 5s. 

When the task is started from `setup()`, it takes ~2s to crash and the ESP reboots.

Any idea? Or any working sample code?

I'll try to extract a failing demo tonight.

0xPIT, 2015-06-08:
BTW: thanks for your work and the support; I've donated some money.

igrr, 2015-06-08:
Okay, how do you pause for 5s? If you use delay() function for that, then it won't work — delay works only from the main task.

You may create a timer which will start your task after some delay:
```C++
#define MY_TASK_PRIORITY 0
#define MY_QUEUE_SIZE    1
static os_event_t g_my_queue[MY_QUEUE_SIZE];
volatile uint32_t g_value = 0;
static os_timer_t task_start_timer;

void my_task(os_event_t *) {
    ++g_value;
    // start the task again in 5 sec
    os_timer_arm(&task_start_timer, 5000 /*ms*/, 0 /*once*/);
}

// this is the timer callback function
void run_task(void*) {
   system_os_post(MY_TASK_PRIORITY, 0, 0);
}

void setup() {
  // create a task
  system_os_task(my_task,
    MY_TASK_PRIORITY, g_my_queue,
    MY_QUEUE_SIZE);

  // create a timer
  os_timer_setfn(&task_start_timer, (os_timer_func_t*) &run_task, 0);

  // start the task for the first time
  run_task(0);
}
```

closed 2015-07-30 by igrr