esp8266/Arduino #400: Possible to use APA102/DotStar?

iamthemadz, 2015-06-08:
I tried to get some APA102 LED's running with the Adafruit DotStar Arduino library. I tried it with an Arduino and the library works with these no problem. They are addressed over SPI. I set the data pin to 9 / gpio14 and clock to 10 / gpio12. For some reason the chip just keeps resetting. Has anyone else had any luck with these LED's?

Makuna, 2015-06-08:
Are you using IO voltage level converters?

iamthemadz, 2015-06-08:
Not yet, the I have some components coming in, so that I can make the logic shifter. However, in this case, the esp module is rebooting itself, which I dont think is normal.

igrr, 2015-06-08:
My guess is that somewhere in this library there is a loop like
```while(true) { wait for something; } ```
which never happens to return. This leads to the watchdog reset.

Regarding SPI, the pins are 12, 13, 14, 15.

iamthemadz, 2015-06-08:
Ok, this is the sketch I am using. It is just the basic strand test. Also I used this page to get the pin number; 

https://github.com/esp8266/esp8266-wiki/wiki/Pin-definition

Is the pin number in the IDE referring to the GPIO number or the physical pin number as laid out in that datasheet?

```
#include <Adafruit_DotStar.h>
#include <SPI.h>

#define NUMPIXELS 8

#define DATAPIN    13 // GPIO15 - MISO
#define CLOCKPIN   9 // GPIO14 - CLK

Adafruit_DotStar strip = Adafruit_DotStar(NUMPIXELS, DATAPIN, CLOCKPIN);

void setup() {
  strip.begin();
  strip.show();
}

int      head  = 0, tail = -10;
uint32_t color = 0xFF0000;

void loop() {

  strip.setPixelColor(head, color);
  strip.setPixelColor(tail, 0);
  strip.show();
  delay(20);

  if(++head >= NUMPIXELS) {
    head = 0;
    if((color >>= 8) == 0)
      color = 0xFF0000;
  }
  if(++tail >= NUMPIXELS) tail = 0;
}
```

Makuna, 2015-06-09:
I took a quick glance at their library, if you supply the pin, they use software bit banging.  All the loops I spotted were for walking pixels, and bytes within pixels.

iamthemadz, 2015-06-09:
Whats the best way to debug this? No matter how I tinker, it just keep resetting every couple of seconds. When I get home, I am going to probe the SPI pins to see if it is actually even doing anything.

Makuna, 2015-06-10:
Did you notice that you start the tail at (-10) and not at zero?  What does 16bit signed -1 get turned into when its converted to an 8bit unsigned when you first call setPixelColor()?  

Start by removing code until the problem goes away.  Comment out all the code associated with Tail.  Still happens? Comment out everything in the Loop().  Still happens?  Comment out everything in Setup().  Still Happens?

iamthemadz, 2015-06-10:
I noticed it there. I am not sure why their example has it. I noticed that when working with these lights the last led often doesnt update on time, so I assumed they use a phantom pixel for the tail.

iamthemadz, 2015-06-10:
Actually I may have had a derp moment. I thought the pins were numbered based on their physical pin number on the chip so I was attempting to init SPI using ridiculous pin numbers. Since I just started using this IDE I was unaware that the GPIO # is what is mapped to pin # ie: GPIO13 = pin 13. 

Furthermore, I realized that there are already constants defined for the SPI pins when you include the SPI library (SS,MOSI,MISO,SCK in that order).

The sketch appears to be running without rebooting so it should be good. However Ill have to wait until later to test the lights.

AndyKirby, 2015-06-10:
Aparently you need an end frame, not sop much an end pixel. The length
of the end frame is related to the length of your strip I think.

https://cpldcpu.wordpress.com/2014/11/30/understanding-the-apa102-superled/

Explains it.

Cheers

Kirbs

iamthemadz, 2015-06-10:
Alright,
The issue was definitely because I was using the wrong pin numbers. Got home, threw the module on to a breadboard, started it without a level shifter (out of curiosity) which didnt work (was not surprised), threw in the level shifter and started it up and it works perfectly.

closed 2015-06-10 by iamthemadz