esp8266/Arduino #401: pgmspace.h functions won't compile/link in a sketch

tdicola, 2015-06-08:
It looks like there's a small issue with linking the pgmspace.h functions (flash memory routines like strlen_P, strcpy_P, etc.).  For example try compiling the sketch below:
```
const char test[] PROGMEM = "This is a test flash string.";

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  Serial.print("Test length: ");
  Serial.println(strlen_P(test));
}

void loop() {
  delay(1000);
}
```
You will see it fail with an error that it can't find the functions when linking:
```
esp8266_string_test.cpp.o:(.text+0x14): undefined reference to `strnlen_P'
esp8266_string_test.cpp.o: In function `setup':
esp8266_string_test.cpp:(.text+0x3d): undefined reference to `strnlen_P'
collect2: error: ld returned 1 exit status
Error compiling.
```
Looking more into it it looks like the pgmspace.h/cpp functions are compiled with G++ as C++ code but don't have an extern "C" wrapper to prevent name mangling.  When gcc links the objects with the sketch it can't find the right symbols and fails.

Luckily the fix is easy, just extend the extern "C" in pgmspace.h that's for the headers so that it also covers the function declarations.  I have a small pull request for this that I'll send in a moment.

Makuna, 2015-06-09:
First, minor, but it is suggested that PROGMEM goes between the type and  the name
```
const char PROGMEM  test[] = "This is a test flash string.";
```
Second, I can't repro your problem.  Are you using the latest builds?

tdicola, 2015-06-10:
Yep I'm using the latest version installed from the boards manager (http://arduino.esp8266.com/package_esp8266com_index.json), looks like it's version 1.6.4-673-g8cd3697.  Just curious, what version are you using?

Makuna, 2015-06-10:
Is this the same issue as https://github.com/esp8266/Arduino/issues/343?

1.6.4-758-ga194024, no problems due to https://github.com/Links2004/Arduino/commit/936669e234e997b1d0406af004d1ac56b900d4fb
my 1.6.4-673-g8cd3697 worked due to  manually applying changes those changes to my local build.

673 is getting a little old.

tdicola, 2015-06-10:
Ahh sorry didn't see that earlier issue, yeah that's probably what I'm hitting.  I'll grab the current github repo version and manually install to check if that fixed it.  Thanks!

Makuna, 2015-06-10:
You can also use the staging json file to get the latest using Arduino Board manager.

tdicola, 2015-06-10:
Ah yep looks like it was just an issue with that older build I was using from the package index json.  After building the latest version of the esp8266 branch from github the example code works great without any changes necessary.  I'll close this and the related pull.

Just curious what's the staging json file, is it a link to a more recent build that you could point Arduino's board manager at?  Thanks!

closed 2015-06-10 by tdicola

Makuna, 2015-06-10:
http://arduino.esp8266.com/staging/package_esp8266com_index.json
use at your own risk, while not exactly a nightly build, it has the most up to date build; but as of right now it is getting a old (missing Servo library port).