esp8266/Arduino #404: sscanf broken

cimba007, 2015-06-08:
```
void setup() {
    // put your setup code here, to run once:
    int i, ii;
    char* test;
    sscanf(test, "%x", &ii);
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Throws this:

```
C:\Users\Jinja\AppData\Local\Temp\build662335425252246733.tmp\sketch_jun08b.cpp.o:(.text+0x4): undefined reference to `sscanf'
C:\Users\Jinja\AppData\Local\Temp\build662335425252246733.tmp\sketch_jun08b.cpp.o: In function `setup':
sketch_jun08b.cpp:(.text+0x14): undefined reference to `sscanf'
collect2.exe: error: ld returned 1 exit status
Fehler beim Kompilieren.
```

Pretty straight forward.
Running on 1.6.4-673-g8cd3697

If anybody has a solution this would be very much appreciated

me-no-dev, 2015-06-08:
os_sprintf(test, "%x", &ii);

cimba007, 2015-06-09:
ty, didn't think of these functions. A wrapper would be nice though

closed 2015-06-09 by cimba007

nicjohnston, 2015-06-27:
os_sprintf didn't work for me, the watchdog timer triggered a reboot.
This is the code:
```c++
void setup() {
  Serial.begin(115200);
  Serial.println("starting");
  char *data = "1234";
  int i = 0;
  os_sprintf(data,"%d", &i);
  Serial.println(i);
}
void loop() { }
```
When os_sprintf is replaced with sscanf, the compiler throws a undefined reference error.

a-andreyev, 2015-08-12:
`os_sprintf(...)` is for different purposes!
We should use `siscanf(...)`, but it's not implemented too.