esp8266/Arduino #408: Serial.available and serial.read seems to be broken

hariprasadiit, 2015-06-10:
Serial.available() and Serial.read() are not working with Arduino board manager's recent version,1.6.4-673-g8cd3697.

this is working fine with github version.

simple echo program over serial is not working.ESP doesn't echo anything back.probably this is related to old github version's serial.read reset bug?

wizard23, 2015-06-17:
Serial.available and serial.read work fine. But your sketch is not running at all. I had the same problem till I figured it out: The official arduino version's serial monitor holds the ESP in reset mode via the RTS wire. So either dont connect RTS and reset and manually initiate the bootloader or use the serial monitor of the github version.

hariprasadiit, 2015-06-23:
now its working,thanks.

closed 2015-06-23 by hariprasadiit