esp8266/Arduino #410: Magic sauce for low power mode?

antelder, 2015-06-10:
I came upon an [EspressifSystems demo](https://github.com/EspressifSystems/low_power_voltage_measurement/) for how to do low power sensors. In the [wiki](https://github.com/EspressifSystems/low_power_voltage_measurement/wiki#magic-sauce-enable-hidden-low-power-features) related to that demo it says:

"In order to enable some undocumented low power features of ESP8266, we have added the following codes in the LOW_POWER_SENSOR_DEMO_DIR/app/Makefile, that modifies part of the flash content. ... This magic sauce command, lowers the power consumption of the chip by 30% when it wakes up to perform sensor measurement." 

Also, further on it says: "We have lowered the standby current, by simply turning off the flash device when we are asleep."

I know about the deep sleep support with the Arduino ESP8266 support and how you can wake up with the WifI on or off, but are these two things used in this demo extra features on top of that? And if so could we have them supportted with the ArduinoIDE - a magic sauce command and turning off the flash device while asleep?

ichilton, 2015-06-13:
I'm interested in this also.

In fact, i'm interested how you did the deep sleep stuff also - is it documented anywhere?

antelder, 2015-06-14:
There's a little doc on deep sleep [here](https://github.com/esp8266/Arduino#esp-specific-apis), or googling it gives lots of discussion. This is a simple example:
```
#include <ESP8266WiFi.h>

void setup() {
  Serial.begin(115200);
  Serial.println();

  Serial.println("deep sleeping for 5 seconds...");
  ESP.deepSleep(5000000, WAKE_RFCAL);
  delay(1000); // needed as sleep seems to take a little while to happen
}

void loop() {
  // wont ever get here
}
```

ichilton, 2015-06-23:
Great, thank you!

Did you find any more with regards to your original question?

Thanks,

Ian

shirish47, 2015-12-30:
how to get into standby mode? or power save mode DTIM 1 & DTIM 2 modes in arduino fashion? is there a function for this??

devyte, 2017-10-20:
There is already deepSleep, as well as other power reduction strategies. I do remember a bin-editing approach by Espressif, but it was a long time ago.
Closing this due to age. 
If anyone thinks the bin-editing approach by Espressif has merit to further reduce power over what is available today in the repo via API, please comment here for further investigation.

closed 2017-10-20 by devyte