esp8266/Arduino #413: Set MAC address?

vshymanskyy, 2015-06-11:
I know how to get the MAC address:

    unsigned char mac[6];
    WiFi.macAddress(mac);

Is there any way to set it?
In the AT firmware, there are commands for this, so I think it should be possible:

    AT+CIPSTAMAC=mac
    AT+CIPAPMAC=mac

chadouming, 2015-06-21:
This is not an issue, ask your question in relevant part of the forum : [Forum](http://www.esp8266.com/viewforum.php?f=25&sid=f82e14927b91bb1b964c93878ef5b1c5)

vshymanskyy, 2015-06-21:
Actually it was meant as a feature request, as I didn't find anything like that (I looked into the code).
Sorry if I missed something...

igrr, 2015-06-21:
This isn't implemented currently, but SDK has the ```wifi_set_macaddr``` function to set MAC addresses. There is a limitation though — this function has to be called early at startup, before setup() method is run. So it's not clear how to make this feature available through Arduino APIs. AT firmware saves the new MAC to EEPROM and restart, if I remember correctly.

vshymanskyy, 2015-06-21:
Thanks for a quick reply! Could we do it the same way, so for example it would change the MAC in the EEPROM, to be applied after restart (no need to restart immediately).

sunnyhighway, 2015-07-24:
I don't think wifi_set_macaddr would cause a reboot, because (according to the SDK programming guide) it returns true or false depending on succes or failure to do so.

It would be pointless to return anything at all if it would cause a reboot, would it?

Links2004, 2015-07-25:
with my change it shut work like this:
```cpp
extern "C" void __run_user_rf_pre_init(void) {
    uint8_t mac[] = {0x00, 0x01, 0x02, 0x03, 0x04, 0x05};
    wifi_set_macaddr(SOFTAP_IF, &mac[0]);
}
```

closed 2015-11-27 by Links2004

ikin77, 2015-12-16:
Hi, does anybody knows why I do get this compiling error?

   undefined reference to `wifi_set_macaddr(unsigned char, unsigned char*)'

I've used 

#include <user_interface.h>
....

byte mac[6];
.... // Set mac value....
WiFi.macAddress(mac);
wifi_set_macaddr(STATION_IF, mac);
....

luc-github, 2015-12-16:
did you add ?
```
extern "C" {
#include "user_interface.h"
}
```

ikin77, 2015-12-16:
@luc-github 
Thanks!!!! I just did "#include <user_interface.h>"

Now I can compile :o) Thanks again!!

ikin77, 2015-12-17:
Hi. I can now change MAC address as STA and SOFAP. When I check on my router's page, I can see the MAC address, buy I can see the different devices names as "ESP-XXXX"

[image]

Can it be changed? I do I do it? Thanks again!!

Links2004, 2015-12-17:
add:
```cpp
WiFi.hostname("your name");
```
before ```WiFi.begin(...);```

vogtitec, 2015-12-27:
@ikin77 
Could you please give a more complete example, WHERE in a sketch you put these lines of code into to change the MAC address of the unit? 
byte mac[6];
.... // Set mac value....
WiFi.macAddress(mac);
wifi_set_macaddr(STATION_IF, mac);
....
I tried it in Links2004's example with
extern "C" void __run_user_rf_pre_init(void) ...
but I get a compile error as "previous declaration of 'void __run_user_rf_pre_init()' with 'C++' linkage" ...

Thanks a lot!

ikin77, 2015-12-27:
Hi. Tomorrow morning I'll post the sample code.

Best regards.

ikin77, 2015-12-28:
Hi. Herer's my code:

     // Set as SoftAP and client
      WiFi.mode(WIFI_AP_STA);       
      
     // Change MAC 
     byte *mac = (byte*)malloc(6);
     WiFi.macAddress(mac);
     mac[0] = 0x19;
     mac[1] = 0x14;
     mac[2] = 0xCA;
     wifi_set_macaddr(STATION_IF, mac);
     mac[2] = 0xCB;
      wifi_set_macaddr(SOFTAP_IF, mac);
      free(mac);

In my sample code I read the original MAC and I change only some bytes. Yo can change others or all or them.

Best regards,

vogtitec, 2015-12-28:
Thank you very much for your example!
By the way, did you succeed in changing the hostname as proposed by Links2004 ? I tried it but nothing seems to be changed ?!

ikin77, 2015-12-29:
I couldn't change the hostnamed, the suggested solution doesn't seems to make any difference. Maybe were not using the function at the right moment, I don't know :o/

Links2004, 2015-12-29:
you need to call it before the DHCP request is made.
may direct after  setup.
```cpp
setup() {
    Serial.begin(115200);
    WiFi.hostname("your_name");
    //.......
}
```

Duality4Y, 2015-12-29:
for setting a hostname you need to run a dhcp server somewhere.

Links2004, 2015-12-29:
@Duality4Y I think you mean:
the DHCP server needs to "add" the names to the DNS server.
the DHCP server itself does not response to a DNS request ;)

Duality4Y, 2015-12-29:
yes sorry that is correct.

vshymanskyy, 2016-12-26:
It looks it got broken at some point...
```
extern "C" {
  #include "user_interface.h"
}

void initVariant() {
  wifi_set_macaddr(STATION_IF, &mac[0]);
}
...
...
  Serial.println(WiFi.macAddress());
```
And I get the original MAC of the module. The router also shows the original MAC.

yoursunny, 2017-07-17:
I've posted a complete example of changing MAC and hostname in my blog:
<https://yoursunny.com/t/2017/change-ESP8266-MAC/>

Setting MAC and hostname can be placed in `setup()` function before `WiFi.begin`. No need to `initVariant`.

IMAN4K, 2019-01-29:
`main.ino:`
```c++
extern "C" void preinit() {
	uint8_t sta_mac[] = { 0xb4, 0xe6, 0x2d, 0x44, 0x86, 0xad };
	wifi_set_opmode(STATIONAP_MODE);
	wifi_set_macaddr(STATION_IF, sta_mac);
}
```
Worked in `core 2.4.2 + ESP12E`

sepp89117, 2019-06-06:
extern "C" void preinit() {
  uint8_t ap_mac[] = {0xD8, 0x96, 0x85, 0x03, 0x04, 0x05};
  wifi_set_opmode(SOFTAP_MODE);
  wifi_set_macaddr(SOFTAP_IF, ap_mac);
}

Worked for me in core 2.5.2 + ESP-12E for SoftAP

happytm, 2019-08-31:
Is it possible to dynamically change mac bytes in code by replacing byte with temperature,humidity and pressure sensor values  read from BME280 that way Raspberry running monitor mode can capture these sensor values?

Thanks

sepp89117, 2019-08-31:
hi, why do you want to do that? 
Please describe your project.

happytm, 2019-08-31:
Basically I am trying to create adhoc network of esp devices (battery powered) to send sensor values to raspberry pi for further processing these value.I donot want to use time consuming TCP protocol to save power.My goal is to have esp device wake up from deep sleep every 15 minutes, read BME280 sensor, dump sensor values into fake MAC id ,do probe request and go back to sleep.On other side raspberry pi is always on and receive this probe request,save these sensor data in database and  publish values via MQTT.

Thanks

sepp89117, 2019-08-31:
In that case I would have used a 433Mhz interface.  With a transmitter in the Arduino transfer the data and listen with the Pi simply.
But if you want to stay with your hardware setup, did you try the code above?
Try wifi_set_macaddr in your action. Theoretically, it should work.

happytm, 2019-08-31:
I want to go with only ESP device for hardware simplicity.Yes I will try wifi_set_macaddr.

Thanks

happytm, 2019-09-22:
I was able to create ad-hoc network between 2 ESP8266 devices and I can exchange 6 bytes of data between 2 devices reliably in about 200 milliseconds. I am using:
int n = WiFi.scanNetworks(true, true, 7); 
to scan all aps with channel 7 which shows up with 4 AP's in my vicinity.

To reduce the time it takes to scan few AP's (60 millisecponds) I want to scan only one known AP with known SSID. 

Is there any way I can do it ?

Thanks